maximkulkin/esp-homekit-demo #397: no such file or directory

Kunalverma1502, 2020-10-21:
step 4) Build esp-sdk Docker container:
 docker build . -f esp-sdk-dockerfile -t esp-sdk. 

gives the error directory not found.

"" failed to solve with frontend dockerfile.v0: failed to read dockerfile: open /var/lib/docker/tmp/buildkit-mount074482500/esp-sdk-dockerfile: no such file or directory ""

can anyone also explain step 1,2,3 in more detailed way!!

maximkulkin, 2020-10-21:
Did you create esp-sdk-dockerfile in current directory (step 2) ?

Kunalverma1502, 2020-10-22:
yes sir , 
I followed the steps please mark me if I am wrong .

firstly I installed python, docker,pip.

step 1) Create an empty directory and change into it.

so I did 
- mkdir new
- cd new

step 2) Create a file esp-sdk-dockerfile with following content:

mkdir esp-sdk-dockerfile

I don't know how to add the below text in file

" FROM ubuntu:16.04 as builder
// ........................//
COPY --from=builder /build/esp-open-sdk/xtensa-lx106-elf /opt/xtensa-lx106-elf
ENV PATH /opt/xtensa-lx106-elf/bin:$PATH
"

step 3) Create a file esp-rtos-dockerfile with following content:

mkdir esp-rtos-dockerfile

I don't know how to add the below text in file

"FROM ubuntu:16.04 as builder
//......................//
ENV SDK_PATH /opt/esp-open-rtos"

step 4) Build esp-sdk Docker container:

gives the error directory not found.

"" failed to solve with frontend dockerfile.v0: failed to read dockerfile: open /var/lib/docker/tmp/buildkit-mount074482500/esp-sdk-dockerfile: no such file or directory ""

please help me
should I use vsc , atom for this, If yes the can you please explain how ?

Alr2303, 2020-10-22:
With ref to https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP8266-(Docker)
use these lines 
1. $ mkdir new
2. $ cd new
3. $ nano esp-sdk-dockerfile
4. paste this in the above file (Ctrl+Shift+V):

FROM ubuntu:16.04 as builder

RUN groupadd -g 1000 docker && useradd docker -u 1000 -g 1000 -s /bin/bash --no-create-home
RUN mkdir /build && chown docker:docker /build

RUN apt-get update && apt-get install -y \
  make unrar-free autoconf automake libtool gcc g++ gperf \
  flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial \
  sed git unzip bash help2man wget bzip2 libtool-bin

RUN su docker -c " \
    git clone --recursive https://github.com/pfalcon/esp-open-sdk.git /build/esp-open-sdk ; \
    cd /build/esp-open-sdk ; \
    make STANDALONE=n ; \
"

FROM ubuntu:16.04

RUN apt-get update && apt-get install -y make python python-serial

COPY --from=builder /build/esp-open-sdk/xtensa-lx106-elf /opt/xtensa-lx106-elf
ENV PATH /opt/xtensa-lx106-elf/bin:$PATH

5. save the file

6. check if python 2, pip2 and python-serial is installed

7. $ mkdir esp-rtos-dockerfile

8. paste this in the above file (Ctrl+Shift+V):

FROM ubuntu:16.04 as builder

RUN apt-get update && apt-get install -y git

RUN git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos

FROM esp-sdk:latest

COPY --from=builder /opt/esp-open-rtos /opt/esp-open-rtos

ENV SDK_PATH /opt/esp-open-rtos

9. save the file

10. $ docker build . -f esp-sdk-dockerfile -t esp-sdk
11. $ docker build . -f esp-rtos-dockerfile -t esp-rtos
12. $ git clone --recursive https://github.com/SuperHouse/esp-open-rtos.git
13. $ pip install esptool
14. $ git clone --recursive https://github.com/maximkulkin/esp-homekit-demo.git
15. $ export SDK_PATH="$(pwd)/esp-open-rtos"
16. $ export ESPPORT=/dev/tty.SLAB_USBtoUART
17. $ mv wifi.h.sample wifi.h
18. $ nano wifi.h 
19. enter the wifi ssid and password and save the file
20. $ cd esp-homekit-demo
21. $ docker run -it --rm -v "$(pwd)":/project -w /project esp-rtos make -C examples/sonoff_basic all
22 $ make -C examples/sonoff_basic flash monitor

This should do the work!

closed 2020-11-07 by maximkulkin