maximkulkin/esp-homekit-demo #398: Linking Error - File not recognized

nikomaxi, 2020-10-23:
Following error while compiling the example button.

./build/wolfssl.a: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
/opt/esp-open-rtos/common.mk:232: recipe for target 'build/button.out' failed
make: *** [build/button.out] Error 1
make: Leaving directory '/project/examples/button'

I used the Docker version.

maximkulkin, 2020-10-24:
Try doing clean rebuild. I've seen such errors, but I do not remember what fixes that. Make sure you do not overwrite LDFLAGS anywhere.

nikomaxi, 2020-10-25:
I have done a clean rebuild but it does show the exact same error. I am using your docker version. So I deleted my docker container and cache from docker. I rebuild the docker container and this error does come again.

maximkulkin, 2020-10-26:
That's not what I meant. Do `make clean` first, then rebuild the firmware.

nikomaxi, 2020-11-11:
I could only solve this problem by reinstalling my OS. 
So I can close this issue.

closed 2020-11-11 by nikomaxi