maximkulkin/esp-homekit-demo #399: Unable to find image 'esp-rtos:latest' locally

Kunalverma1502, 2020-11-03:
ALL THE STEPS WORKS FINE TILL STEP 12 

When i tried building the example by writing -  

docker run -it --rm -v "$(pwd)":/project -w /project esp-rtos make -C examples/sonoff_basic all

it showed the error -  

Unable to find image 'esp-rtos:latest' locally
docker: Error response from daemon: pull access denied for esp-rtos, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.

maximkulkin, 2020-11-04:
Does not seem like you have esp-rtos image. Did you do step 5? What's the output of `docker images | grep esp-rtos` ?

kjellxvx, 2020-11-04:
I have the same issue. Step 5 shows the following output: 
Sending build context to Docker daemon    326MB
Step 1/6 : FROM ubuntu:16.04 as builder
 ---> dfeff22e96ae
Step 2/6 : RUN apt-get update && apt-get install -y git
 ---> Using cache
 ---> d2cac5f734ff
Step 3/6 : RUN git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos
 ---> Running in 62642e3e6301
Cloning into '/opt/esp-open-rtos'...
Submodule 'bootloader/rboot' (https://github.com/raburton/rboot.git) registered for path 'bootloader/rboot'
Submodule 'extras/bearssl/BearSSL' (https://www.bearssl.org/git/BearSSL) registered for path 'extras/bearssl/BearSSL'
Submodule 'extras/crc_generic/crc_lib' (https://github.com/Zaltora/crc_generic_lib.git) registered for path 'extras/crc_generic/crc_lib'
Submodule 'extras/http-parser/http-parser' (https://github.com/nodejs/http-parser) registered for path 'extras/http-parser/http-parser'
Submodule 'extras/jsmn/jsmn' (https://github.com/zserge/jsmn.git) registered for path 'extras/jsmn/jsmn'
Submodule 'extras/libesphttpd/libesphttpd' (https://github.com/nochkin/libesphttpd) registered for path 'extras/libesphttpd/libesphttpd'
Submodule 'extras/mbedtls/mbedtls' (https://github.com/ARMmbed/mbedtls.git) registered for path 'extras/mbedtls/mbedtls'
Submodule 'extras/multipwm' (https://github.com/nochkin/multipwm) registered for path 'extras/multipwm'
Submodule 'extras/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'extras/spiffs/spiffs'
Submodule 'lvgl/lv_drivers' (https://github.com/littlevgl/lv_drivers.git) registered for path 'lvgl/lv_drivers'
Submodule 'lvgl/lv_examples' (https://github.com/littlevgl/lv_examples.git) registered for path 'lvgl/lv_examples'
Submodule 'lvgl/lvgl' (https://github.com/littlevgl/lvgl.git) registered for path 'lvgl/lvgl'
Submodule 'lwip/lwip' (https://github.com/ourairquality/lwip.git) registered for path 'lwip/lwip'
Submodule 'tests/fs-test' (https://github.com/sheinz/fs-test) registered for path 'tests/fs-test'
Submodule 'tests/unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'tests/unity'
Cloning into 'bootloader/rboot'...
Submodule path 'bootloader/rboot': checked out '3b067a8686a07474c5f722953d3b2f0e71db681f'
Cloning into 'extras/bearssl/BearSSL'...
Submodule path 'extras/bearssl/BearSSL': checked out '2b738493bd16d57fdb12d38d03631981370259be'
Cloning into 'extras/crc_generic/crc_lib'...
Submodule path 'extras/crc_generic/crc_lib': checked out 'a97013c72f686735889b7a0908bbbf15a104f7d0'
Cloning into 'extras/http-parser/http-parser'...
Submodule path 'extras/http-parser/http-parser': checked out '335850f6b868d3411968cbf5a4d59fe619dee36f'
Cloning into 'extras/jsmn/jsmn'...
Submodule path 'extras/jsmn/jsmn': checked out 'bbc6755fce14c713f9bb4ba47c688d15efc1394b'
Cloning into 'extras/libesphttpd/libesphttpd'...
Submodule path 'extras/libesphttpd/libesphttpd': checked out '30472301589fb0c4eae792884cafc92651a959e3'
Submodule 'lib/heatshrink' (https://github.com/atomicobject/heatshrink.git) registered for path 'lib/heatshrink'
Cloning into 'lib/heatshrink'...
Submodule path 'extras/libesphttpd/libesphttpd/lib/heatshrink': checked out '7d419e1fa4830d0b919b9b6a91fe2fb786cf3280'
Cloning into 'extras/mbedtls/mbedtls'...
Submodule path 'extras/mbedtls/mbedtls': checked out '1bf6123fca97a9a35c2f403ab0c96495f9580db6'
Cloning into 'extras/multipwm'...
Submodule path 'extras/multipwm': checked out '44ecea5757bee0112c65f33eadb44e91a8923535'
Cloning into 'extras/spiffs/spiffs'...
Submodule path 'extras/spiffs/spiffs': checked out '39937743fbbec4b82308ee08332bf9180408d23b'
Cloning into 'lvgl/lv_drivers'...
fatal: reference is not a tree: 8fe0804ceb875193a37106a2e25d07c106650a86
Cloning into 'lvgl/lv_examples'...
Submodule path 'lvgl/lv_examples': checked out 'c3afbc9819f5492a7fe7e7e18a37df95fb60f449'
Cloning into 'lvgl/lvgl'...
Submodule path 'lvgl/lvgl': checked out '41695bf9ac758ae3276424e06308e1bb89c4c019'
Cloning into 'lwip/lwip'...
Submodule path 'lwip/lwip': checked out 'd74e9ad2f7c9db996fb398cd41bf59ef463ae6fe'
Cloning into 'tests/fs-test'...
Submodule path 'tests/fs-test': checked out 'e531bc0d4f75887e5f0e081aae3efbf4a50e2f54'
Cloning into 'tests/unity'...
Submodule path 'tests/unity': checked out 'bbf2fe3a934f96cd00693841247a689e57a17b0d'
Unable to checkout '8fe0804ceb875193a37106a2e25d07c106650a86' in submodule path 'lvgl/lv_drivers'
The command '/bin/sh -c git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos' returned a non-zero code: 1

kjellxvx, 2020-11-04:
this does fix the issue: https://github.com/maximkulkin/esp-homekit-demo/issues/389#issuecomment-697274335

Kunalverma1502, 2020-11-05:
After running the step 5)

5. Build esp-rtos Docker container:

docker build . -f esp-rtos-dockerfile -t esp-rtos

IT SAYS !!

[+] Building 131.3s (8/9)                                                       
 => [internal] load build definition from esp-rtos-dockerfile              0.0s
 => => transferring dockerfile: 336B                                       0.0s
 => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0s
 => [internal] load metadata for docker.io/library/esp-sdk:latest          0.0s
 => [internal] load metadata for docker.io/library/ubuntu:16.04            1.1s
 => CACHED [stage-1 1/2] FROM docker.io/library/esp-sdk:latest             0.0s
 => [builder 1/3] FROM docker.io/library/ubuntu:16.04@sha256:bb69f1a2b6c8  0.0s
 => CACHED [builder 2/3] RUN apt-get update && apt-get install -y git      0.0s
 => ERROR [builder 3/3] RUN git clone --recursive https://github.com/Su  130.1s
------                                                                          
 > [builder 3/3] RUN git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos:                                                  
#8 0.632 Cloning into '/opt/esp-open-rtos'...                                   
#8 10.84 Submodule 'bootloader/rboot' (https://github.com/raburton/rboot.git) registered for path 'bootloader/rboot'                                            
#8 10.86 Submodule 'extras/bearssl/BearSSL' (https://www.bearssl.org/git/BearSSL) registered for path 'extras/bearssl/BearSSL'
#8 10.89 Submodule 'extras/crc_generic/crc_lib' (https://github.com/Zaltora/crc_generic_lib.git) registered for path 'extras/crc_generic/crc_lib'
#8 10.91 Submodule 'extras/http-parser/http-parser' (https://github.com/nodejs/http-parser) registered for path 'extras/http-parser/http-parser'
#8 10.93 Submodule 'extras/jsmn/jsmn' (https://github.com/zserge/jsmn.git) registered for path 'extras/jsmn/jsmn'
#8 10.95 Submodule 'extras/libesphttpd/libesphttpd' (https://github.com/nochkin/libesphttpd) registered for path 'extras/libesphttpd/libesphttpd'
#8 10.98 Submodule 'extras/mbedtls/mbedtls' (https://github.com/ARMmbed/mbedtls.git) registered for path 'extras/mbedtls/mbedtls'
#8 11.00 Submodule 'extras/multipwm' (https://github.com/nochkin/multipwm) registered for path 'extras/multipwm'
#8 11.02 Submodule 'extras/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'extras/spiffs/spiffs'
#8 11.05 Submodule 'lvgl/lv_drivers' (https://github.com/littlevgl/lv_drivers.git) registered for path 'lvgl/lv_drivers'
#8 11.07 Submodule 'lvgl/lv_examples' (https://github.com/littlevgl/lv_examples.git) registered for path 'lvgl/lv_examples'
#8 11.09 Submodule 'lvgl/lvgl' (https://github.com/littlevgl/lvgl.git) registered for path 'lvgl/lvgl'
#8 11.11 Submodule 'lwip/lwip' (https://github.com/ourairquality/lwip.git) registered for path 'lwip/lwip'
#8 11.14 Submodule 'tests/fs-test' (https://github.com/sheinz/fs-test) registered for path 'tests/fs-test'
#8 11.16 Submodule 'tests/unity' (https://github.com/ThrowTheSwitch/Unity.git) registered for path 'tests/unity'
#8 11.18 Cloning into 'bootloader/rboot'...
#8 13.96 Submodule path 'bootloader/rboot': checked out '3b067a8686a07474c5f722953d3b2f0e71db681f'
#8 13.98 Cloning into 'extras/bearssl/BearSSL'...
#8 18.85 Submodule path 'extras/bearssl/BearSSL': checked out '2b738493bd16d57fdb12d38d03631981370259be'
#8 18.87 Cloning into 'extras/crc_generic/crc_lib'...
#8 20.75 Submodule path 'extras/crc_generic/crc_lib': checked out 'a97013c72f686735889b7a0908bbbf15a104f7d0'
#8 20.77 Cloning into 'extras/http-parser/http-parser'...
#8 23.73 Submodule path 'extras/http-parser/http-parser': checked out '335850f6b868d3411968cbf5a4d59fe619dee36f'
#8 23.76 Cloning into 'extras/jsmn/jsmn'...
#8 26.20 Submodule path 'extras/jsmn/jsmn': checked out 'bbc6755fce14c713f9bb4ba47c688d15efc1394b'
#8 26.24 Cloning into 'extras/libesphttpd/libesphttpd'...
#8 30.34 Submodule path 'extras/libesphttpd/libesphttpd': checked out '30472301589fb0c4eae792884cafc92651a959e3'
#8 30.36 Submodule 'lib/heatshrink' (https://github.com/atomicobject/heatshrink.git) registered for path 'lib/heatshrink'
#8 30.39 Cloning into 'lib/heatshrink'...
#8 32.70 Submodule path 'extras/libesphttpd/libesphttpd/lib/heatshrink': checked out '7d419e1fa4830d0b919b9b6a91fe2fb786cf3280'
#8 32.73 Cloning into 'extras/mbedtls/mbedtls'...
#8 64.89 Submodule path 'extras/mbedtls/mbedtls': checked out '1bf6123fca97a9a35c2f403ab0c96495f9580db6'
#8 64.92 Cloning into 'extras/multipwm'...
#8 66.78 Submodule path 'extras/multipwm': checked out '44ecea5757bee0112c65f33eadb44e91a8923535'
#8 66.81 Cloning into 'extras/spiffs/spiffs'...
#8 69.70 Submodule path 'extras/spiffs/spiffs': checked out '39937743fbbec4b82308ee08332bf9180408d23b'
#8 69.73 Cloning into 'lvgl/lv_drivers'...
#8 73.17 fatal: reference is not a tree: 8fe0804ceb875193a37106a2e25d07c106650a86
#8 73.19 Cloning into 'lvgl/lv_examples'...
#8 97.68 Submodule path 'lvgl/lv_examples': checked out 'c3afbc9819f5492a7fe7e7e18a37df95fb60f449'
#8 97.71 Cloning into 'lvgl/lvgl'...
#8 113.0 Submodule path 'lvgl/lvgl': checked out '41695bf9ac758ae3276424e06308e1bb89c4c019'
#8 113.0 Cloning into 'lwip/lwip'...
#8 122.3 Submodule path 'lwip/lwip': checked out 'd74e9ad2f7c9db996fb398cd41bf59ef463ae6fe'
#8 122.3 Cloning into 'tests/fs-test'...
[... cut ...] 5f0e081aae3efbf4a50e2f54'
#8 124.1 Cloning into 'tests/unity'...
#8 130.0 Submodule path 'tests/unity': checked out 'bbf2fe3a934f96cd00693841247a689e57a17b0d'
#8 130.0 Unable to checkout '8fe0804ceb875193a37106a2e25d07c106650a86' in submodule path 'lvgl/lv_drivers'
------
failed to solve with frontend dockerfile.v0: failed to build LLB: executor failed running [/bin/sh -c git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos]: runc did not terminate sucessfully

Kunalverma1502, 2020-11-05:
The following changes to the Dockerfiles worked for me:

esp-rtos-dockerfile

FROM ubuntu:18.04 as builder

RUN apt-get update && apt-get install -y git

RUN git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos

FROM esp-sdk:latest

COPY --from=builder /opt/esp-open-rtos /opt/esp-open-rtos

ENV SDK_PATH /opt/esp-open-rtos
Essentially, updated to Ubuntu 18.04

cc @JohnWickham @maximkulkin

JohnWickham, 2020-11-05:
Thanks @Kunalverma1502, unfortunately that didn't solve this issue for me.

`docker images | grep esp-rtos` outputs:

    esp-rtos        latest        2e1c7b24f94d        2 minutes ago        782MB

But I still get the original “Unable to find image 'esp-rtos:latest' locally” error when trying to build.

Kunalverma1502, 2020-11-05:
JUST FOLLOW THESE STEPS , START WITH A FRESH BUILD.

STEP 1) Create an empty directory and change into it.

STEP 1.2) $ mkdir new

STEP 1.3) $  cd new

--------------------------------------------------------------------------------------------------------

STEP 2)  Create a file esp-sdk-dockerfile with following content:

STEP 2.1) $ nano esp-sdk-dockerfile

STEP 2.2) THEN PASTE THE FOLLOWING GIVEN BELOW AND SAVE CHANGES.

--------------------------------------------------------------------------------------------------------

FROM ubuntu:16.04 as builder

RUN groupadd -g 1000 docker && useradd docker -u 1000 -g 1000 -s /bin/bash --no-create-home
RUN mkdir /build && chown docker:docker /build

RUN apt-get update && apt-get install -y \
  make unrar-free autoconf automake libtool gcc g++ gperf \
  flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial \
  sed git unzip bash help2man wget bzip2 libtool-bin

RUN su docker -c " \
    git clone --recursive https://github.com/pfalcon/esp-open-sdk.git /build/esp-open-sdk ; \
    cd /build/esp-open-sdk ; \
    make STANDALONE=n ; \
"

FROM ubuntu:16.04

RUN apt-get update && apt-get install -y make python python-serial

COPY --from=builder /build/esp-open-sdk/xtensa-lx106-elf /opt/xtensa-lx106-elf
ENV PATH /opt/xtensa-lx106-elf/bin:$PATH

--------------------------------------------------------------------------------------------------------

STEP 3) Create a file esp-rtos-dockerfile with following content:

STEP 3.1) $ nano esp-rtos-dockerfile

STEP 3.2) THEN PASTE THE FOLLOWING GIVEN BELOW AND SAVE CHANGES.

--------------------------------------------------------------------------------------------------------

FROM ubuntu:18.04 as builder

RUN apt-get update && apt-get install -y git

RUN git clone --recursive https://github.com/Superhouse/esp-open-rtos.git /opt/esp-open-rtos

FROM esp-sdk:latest

COPY --from=builder /opt/esp-open-rtos /opt/esp-open-rtos

ENV SDK_PATH /opt/esp-open-rtos

--------------------------------------------------------------------------------------------------------

STEP 4) Build esp-sdk Docker container:

STEP 4.1) $ docker build . -f esp-sdk-dockerfile -t esp-sdk

--------------------------------------------------------------------------------------------------------

STEP 5) Build esp-rtos Docker container:

STEP 5.1) $ docker build . -f esp-rtos-dockerfile -t esp-rtos

--------------------------------------------------------------------------------------------------------

If all this steps works without any issue then continue the rest steps given on the main page 

if anything like failed appears on the cmd , terminal then copy and paste it in the comment for help.

cc @JohnWickham @maximkulkin

maximkulkin, 2020-11-07:
Hope that solved the problem

closed 2020-11-07 by maximkulkin