maximkulkin/esp-homekit-demo #400: Unable to compile examples/magic_home_strip

EG-Julien, 2020-11-04:
Trying to compile magic_home_strip example but got this error ... Tried many different things but nothing worked atm. Any ideas ?

(I have compiled temperature_sensor example without any problem one minute before trying this example)

```
make: Entering directory '/home/ju/Documents/esp-homekit/esp-homekit-demo/examples/magic_home_strip'
CC /home/ju/Documents/esp-homekit/esp-homekit-demo/components/esp8266-open-rtos/wifi_config/src/wifi_config.c
/home/ju/Documents/esp-homekit/esp-homekit-demo/components/esp8266-open-rtos/wifi_config/src/wifi_config.c: In function 'wifi_config_server_on_settings':
/home/ju/Documents/esp-homekit/esp-homekit-demo/components/esp8266-open-rtos/wifi_config/src/wifi_config.c:217:31: error: 'html_settings_header' undeclared (first use in this function)
     client_send_chunk(client, html_settings_header);
                               ^
/home/ju/Documents/esp-homekit/esp-homekit-demo/components/esp8266-open-rtos/wifi_config/src/wifi_config.c:217:31: note: each undeclared identifier is reported only once for each function it appears in
/home/ju/Documents/esp-homekit/esp-homekit-demo/components/esp8266-open-rtos/wifi_config/src/wifi_config.c:220:31: error: 'html_settings_custom_html' undeclared (first use in this function)
  uint8_t buffer_size = strlen(html_settings_custom_html) + strlen(context->custom_html); 
                               ^
/home/ju/Documents/esp-homekit/esp-homekit-demo/components/esp8266-open-rtos/wifi_config/src/wifi_config.c:227:31: error: 'html_settings_body' undeclared (first use in this function)
     client_send_chunk(client, html_settings_body);
                               ^
/home/ju/Documents/esp-homekit/esp-homekit-demo/components/esp8266-open-rtos/wifi_config/src/wifi_config.c:235:17: error: 'html_network_item' undeclared (first use in this function)
                 html_network_item,
                 ^
/home/ju/Documents/esp-homekit/esp-homekit-demo/components/esp8266-open-rtos/wifi_config/src/wifi_config.c:246:31: error: 'html_settings_footer' undeclared (first use in this function)
     client_send_chunk(client, html_settings_footer);
                               ^
make: *** [/home/ju/Documents/esp-homekit/esp-homekit-demo/components/esp8266-open-rtos/wifi_config/component.mk:10: build/wifi_config//src/wifi_config.o] Error 1
make: Leaving directory '/home/ju/Documents/esp-homekit/esp-homekit-demo/examples/magic_home_strip'

```

closed 2020-11-04 by EG-Julien

burgonyapure, 2020-11-22:
Hi!
I have the same issue, but while trying to recompile the thermostat example.
Could you resolve your problem?

maximkulkin, 2020-11-22:
@burgonyapure Did you use make, cmake or idf.py to compile firmware? Try using `make`

burgonyapure, 2020-11-22:
In the meantime i've got it resolved by deleting the previously created build directory, and after it rebuilt successfully.
Thanks for your quick response

EG-Julien, 2020-11-23:
@burgonyapure make clean solved the problem for me.