maximkulkin/esp-homekit-demo #403: stops working from time to time

kjellxvx, 2020-11-09:
I have the issue, that everything works fine right after starting the device. However, after some time, the device seems to loose connection or it hangs up. A restart fixes the issue but since this is home automation, a restart before using the device is kind of senseless. Anyone else having this problem and know how to fix this?

I am running the Magic Home Strip example code on a MagicHome RGBW controller.

Kristian8606, 2020-11-09:
Try bringing the device closer to your WiFi.

kjellxvx, 2020-11-13:
I can't put it 1m next to my router. That's not the idea behind automation :D 
The devices definitely are in reach and work perfectly right after plugging in. Looks more like a software bug zu me.

maximkulkin, 2020-11-13:
Recently I have a suspicion that a lot of HomeKit instability comes from ESP8266 WiFi stack. If you're building this with esp-open-rtos, try porting it to ESP8266_RTOS_SDK and see if that helps. esp-open-rtos has WiFi stack embedded as a precompiled libraries which haven't been updated for quite some time.

kjellxvx, 2020-11-13:
> Recently I have a suspicion that a lot of HomeKit instability comes from ESP8266 WiFi stack. If you're building this with esp-open-rtos, try porting it to ESP8266_RTOS_SDK and see if that helps. [...]

thanks for the reply! Can you tell me how I can port it to ESP8266_RTOS_SDK? I'm doing everything along your build instructions.

maximkulkin, 2020-11-13:
Look at [ESP8266_RTOS_SDK example ](https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/esp8266-rtos-sdk/led)and make your example look like this one. Install [ESP8266_RTOS_SDK](https://github.com/espressif/ESP8266_RTOS_SDK) according to instructions. Export IDF_PATH environment variable pointing to ESP8266_RTOS_SDK directory and do `make all` (you do not need to build it inside Docker, but you will need to install their toolchain onto host machine).

kjellxvx, 2020-11-14:
When comparing the regular led example and the RTOS_SDK led example the main differences seem to be the on_wifi_ready void, wifi_init void and app_main void. Is that right? Everything else looks the same and should also work the same. I'll try to bring the magic home code in that format and try to make it. Thanks for the explanation so far!

kjellxvx, 2020-11-14:
Installed RTOS_SDK, Exported IDF_PATH, installes toolchain, make all and I get the following error:

In file included from /Users/kjellxvx/Code/docker-homekit/esp-homekit-demo/components/common/homekit/src/port.c:80:
/Users/kjellxvx/esp/ESP8266_RTOS_SDK/components/mdns/include/mdns.h:347:69: error: unknown type name 'ip6_addr_t'; did you mean 'ip_addr_t'?
 esp_err_t mdns_query_aaaa(const char * host_name, uint32_t timeout, ip6_addr_t * addr);
                                                                     ^~~~~~~~~~
                                                                     ip_addr_t
make[1]: *** [src/port.o] Error 1
make: *** [component-homekit-build] Error 2

Building the hello world example in the RTOS_SDK Getting started guide, works without any error. 
Maybe I should open a new issue for building esp8266-rtos-sdk...

mriksman, 2020-11-14:
I have a wiki at https://github.com/mriksman/esp-idf-homekit speaking about my experiences with using this with ESP-IDF and ESP_RTOS_SDK.

closed 2020-12-16 by kjellxvx