maximkulkin/esp-homekit-demo #404: Using MQTT with esp-homekit

sam632, 2020-11-09:
Hi, I am trying to combine MQTT with HomeKit to make a pair of friendship lamps, each with 2 LEDs, one controlled by HomeKit (and a button) and the other controlled by the other persons lamp. The MQTT code works on its own but for some reason when I combine with HomeKit it won't work. The phone will not connect to the device and after it tries for a while, the MQTT connection drops. I tried adding a vTaskSuspend to the on_password function but this doesn't seem to get called. Any help would be much appreciated. (I used maqiatto's free MQTT Broker). 

'''
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <espressif/esp_common.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <string.h>
#include <paho_mqtt_c/MQTTESP8266.h>
#include <paho_mqtt_c/MQTTClient.h>

#include <semphr.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>

#include "wifi.h"

#include "button.h"

#define MQTT_HOST ("maqiatto.com")
#define MQTT_PORT 1883

#define MQTT_USER ("myemail")
#define MQTT_PASS ("mypassword")

SemaphoreHandle_t wifi_alive;
QueueHandle_t publish_queue;
#define PUB_MSG_LEN 2

int led2_state = 0;

// The GPIO pin that is connected to the local LED
const int led1_gpio = 5; //D1
//The GPIO pin that is connected to the externally controlled LED
const int led2_gpio = 4; //D2
//The GPIO pin that is connected to the LED on the button
const int led3_gpio = 14; //D5
// The GPIO pin that is connected to the button on the WEMOS
const int button_gpio = 0; //D3 with internal pullup resistor

// Define callback functions
void switch_on_callback(homekit_characteristic_t *_ch, homekit_value_t on, void *context);
void button_callback(button_event_t event, void* context);
TaskHandle_t MQTTTask;

// Define LED functions
void led1_write(bool on) {
    gpio_write(led1_gpio, on ? 1 : 0);
}
void led2_write(bool on) {
    gpio_write(led2_gpio, on ? 1 : 0);
}
void led3_write(bool on) {
    gpio_write(led3_gpio, on ? 1 : 0);
}

static void  wifi_task(void *pvParameters) {
    
    uint8_t status  = 0;
    uint8_t retries = 30;
    struct sdk_station_config config = {
        .ssid = WIFI_SSID,
        .password = WIFI_PASSWORD,
    };
[... 356 lines cut ...]
    char *name_value = malloc(name_len+1);
    snprintf(name_value, name_len+1, "Friendship Lamp-%02X%02X%02X",
             macaddr[3], macaddr[4], macaddr[5]);
    
    name.value = HOMEKIT_STRING(name_value);
}

void user_init(void) {
    uart_set_baud(0, 115200);

    create_accessory_name();
    
    vSemaphoreCreateBinary(wifi_alive);
    publish_queue = xQueueCreate(3, PUB_MSG_LEN);
    wifi_init();
    
    mqtt_init();
    gpio_init();
    button_init();
    
    homekit_server_init(&config);
   
}

```

lrusnac, 2020-11-10:
I did not read your code, but I do use mqtt with my devices, so you can find some examples of using mqtt here https://github.com/lrusnac/esp-homekit-devices
hope it is useful

sam632, 2020-11-10:
Thank you very much, I will look at this now :)

closed 2020-11-10 by sam632

peros550, 2020-11-10:
@lrusnac  you are the first (to my knowledge) that you have brought together mqtt and esp-homekit. 
How is stability and performance? Are you getting any sort of "no response" status in your devices?

lrusnac, 2020-11-10:
@peros550  I doubt I'm the first, but I have a couple of devices running for 10 months now (based on my examples commits) and it works well so far, haven't noticed any issues so far. 

I even have one esp that is a bridge between mqtt and homekit without ant sensors or actuators attached, instead of running homebridge. Pretty stable, and fast

sam632, 2020-11-10:
Worked a treat, here is my updated code for anyone interested. 

'''c
#include <stdio.h>
#include <espressif/esp_wifi.h>
#include <espressif/esp_sta.h>
#include <espressif/esp_common.h>
#include <esp/uart.h>
#include <esp8266.h>
#include <FreeRTOS.h>
#include <task.h>

#include <string.h>
#include <paho_mqtt_c/MQTTESP8266.h>
#include <paho_mqtt_c/MQTTClient.h>

#include <semphr.h>

#include <homekit/homekit.h>
#include <homekit/characteristics.h>

#include <wifi_config.h>

#include "button.h"

#define MQTT_HOST ("maqiatto.com")
#define MQTT_PORT 1883
#define MQTT_USER ("username")
#define MQTT_PASS ("password")
#define MQTT_CMND_TOPIC "username/topic" //Controls LED 2
#define MQTT_SEND_TOPIC "username/topic" //Sends MQTT to control other lamp

QueueHandle_t publish_queue;
#define PUB_MSG_LEN 2

int led2_state = 0;

// The GPIO pin that is connected to the local LED
const int led1_gpio = 5; //D1
//The GPIO pin that is connected to the externally controlled LED
const int led2_gpio = 4; //D2
//The GPIO pin that is connected to the LED on the button
const int led3_gpio = 14; //D5
// The GPIO pin that is connected to the button on the WEMOS
const int button_gpio = 0; //D3 with internal pullup resistor

// Define callback functions
void switch_on_callback(homekit_characteristic_t *_ch, homekit_value_t on, void *context);
void button_callback(button_event_t event, void* context);

// Define LED functions
void led1_write(bool on) {
    gpio_write(led1_gpio, on ? 1 : 0);
}
void led2_write(bool on) {
    gpio_write(led2_gpio, on ? 1 : 0);
}
void led3_write(bool on) {
    gpio_write(led3_gpio, on ? 1 : 0);
}

// For resetting the wifi
void reset_configuration_task() {
    //Flash the LED first before we start the reset
    for (int i=0; i<3; i++) {
        led3_write(true);
        vTaskDelay(100 / portTICK_PERIOD_MS);
        led3_write(false);
        vTaskDelay(100 / portTICK_PERIOD_MS);
    }
[... 292 lines cut ...]
    sdk_wifi_get_macaddr(STATION_IF, macaddr);
    
    int name_len = snprintf(NULL, 0, "Friendship Lamp-%02X%02X%02X",
                            macaddr[3], macaddr[4], macaddr[5]);
    char *name_value = malloc(name_len+1);
    snprintf(name_value, name_len+1, "Friendship Lamp-%02X%02X%02X",
             macaddr[3], macaddr[4], macaddr[5]);
    
    name.value = HOMEKIT_STRING(name_value);
}

void user_init(void) {
    uart_set_baud(0, 115200);

    create_accessory_name();
    
    publish_queue = xQueueCreate(3, PUB_MSG_LEN);
    wifi_config_init("Friendship Lamps -", NULL, on_wifi_ready);

    gpio_init();
    button_init();
   
}

```