maximkulkin/esp-homekit-demo #407: error: python is missing or unusable

Golovkov, 2020-11-11:
good afternoon!
when executing the command
`docker build . -f esp-sdk-dockerfile -t esp-sdk`
ran into error:
`[ERROR]    configure: error: python is missing or unusable
[ERROR]    make[4]: *** [configure-gdb] Error 1
[ERROR]    make[3]: *** [all] Error 2
[ERROR]  - 
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing cross-gdb'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@120]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6/B - Known issues.txt'
[ERROR]  / 
[ERROR]  (elapsed: 876:12.45)
[876:14] / ct-ng:152: recipe for target 'build' failed
#9 5.260 make[2]: *** [build] Error 2
#9 5.260 make[2]: Leaving directory '/build/esp-open-sdk/crosstool-NG'
#9 5.260 ../Makefile:135: recipe for target '_toolchain' failed
#9 5.260 make[1]: *** [_toolchain] Error 2
#9 5.260 make[1]: Leaving directory '/build/esp-open-sdk/crosstool-NG'
#9 5.260 Makefile:130: recipe for target 'crosstool-NG/.built' failed
#9 5.260 make: *** [crosstool-NG/.built] Error 2
------`
Please tell me how you can solve it?

Golovkov, 2020-11-11:
solution with
`sudo apt-get install python-dev`
does not work and will throw an error

> No available formula or cask with the name "python-dev".

maximkulkin, 2020-11-11:
Not sure how this has happened, python-dev is installed a few steps above in that Dockerfile (assuming you're using Dockerfile from step 2 on [the wiki](https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP8266-(Docker))).

AchimPieters, 2021-02-04:
maybe this will help:[ installation](https://www.studiopieters.nl/esp-homekit-sdk-revised-installation/)