maximkulkin/esp-homekit-demo #408: I can't pair it with homekit

Kristian8606, 2020-11-24:
I try to pair the device but always to no avail

>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 78:2D:23:65:7A:7E
>>> HomeKit: Configuring mDNS
>>> HomeKit: Got new client connection: 55 from 192.168.0.110
>>> HomeKit: [Client 55] Pair Setup Step 1/3
>>> HomeKit: [Client 55] Pair Setup Step 2/3
>>> HomeKit: [Client 55] Pair Setup Step 3/3
>>> HomeKit: Added pairing with EE50190D-642E-4437-B1F7-657F0714E992
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 55] Successfully paired
>>> HomeKit: [Client 55] Closing client connection

after restart

`WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 78:2D:23:65:7A:7E
>>> HomeKit: Found admin pairing with EE50190D-642E-4437-B1F7-657F0714E992, disabling pair setup
>>> HomeKit: Configuring mDNS`

Kristian8606, 2020-11-24:
[image]

maximkulkin, 2020-11-24:
It says "after restart". What restart? You're not supposed to reboot accessory right after pairing.

Kristian8606, 2020-11-24:
after restart

`WiFI ready

HomeKit: Starting server
HomeKit: Using existing accessory ID: 78:2D:23:65:7A:7E
HomeKit: Found admin pairing with EE50190D-642E-4437-B1F7-657F0714E992, disabling pair setup
HomeKit: Configuring mDNS`

Kristian8606, 2020-11-24:
but no accessory is displayed in the home.  When trying to pair again, the accessory was not found because it appears to be paired

Kristian8606, 2020-11-24:
Because of my poor English, my question may have confused you, sorry.  When I try to pair I get a `HomeKit: [Client 55] Closing client connection ` and the error in the photo above.  Then I restart esp and try to pair again but it behaves as if it is paired

maximkulkin, 2020-11-24:
Disconnection after pair setup is a normal procedure. Controller then reconnects, goes through pair verification and then downloads accessory definition and shows what kind of services accessory provides on the UI. From your log I can't tell what happens after pair setup.

peros550, 2020-11-24:
if I got it right , Kristian says that accessory paired in logs, but never in home iOS app.

Kristian8606, 2020-11-24:
this is all available information and according to it the device seems to be paired but in fact the device is not displayed in the home application due to the photo error

Kristian8606, 2020-11-24:
I noticed that the size of led.bin is 681 kb I think this may be a problem?

AramVartanyan, 2020-11-24:
This issue could be triggered by values out of boundaries or improper characteristics defined. I have never received this message. 🤔

Kristian8606, 2020-11-24:
just following the example i haven't changed anything

Kristian8606, 2020-11-25:
I forgot to specify that I use led example ESP8266_RTOS_SDK

Kristian8606, 2020-11-28:
What causes this error?
I get it often.
`!!! HomeKit: [Client 4] Failed to compute SRP shared secret (code -2)`

maximkulkin, 2020-11-28:
This is out of memory error.

peros550, 2020-11-28:
I get this error also, right after setting up the wifi. Solution seems to be to hard reset the board and try again. It looks the wifi config procedure eats some memory which isn’t freed?

maximkulkin, 2020-11-28:
@peros550 I do not know what is exactly in your case, but wifi_config can cause memory fragmentation (because you run homekit server when wifi_config hasn't fully stopped yet), so it goes like  ( <wifi_config> <homekit_server> <free ram> ), then wifi_config stops and you get ( <free ram> <homekit_server> <free ram>), so memory is fragmented and there might not be enough memory chunks.

peros550, 2020-11-28:
Thanks Maxim. No worries it's not a big issue on my case since a reboot clears the problem. 
Any idea how we could calculate what kind of memory its RTOS task consumes in design time?

maximkulkin, 2020-11-30:
@peros550 Yes, you can. Basically, it's the stack size you allocate for your task + some fixed overhead ( + everything your task allocates while running ).

tosie, 2020-12-04:
@Kristian8606 In the past I have seen this problem when not using the newest git submodules. Try to update all submodules, rebuild everything, erase the flash and flash again (maybe some of these steps are not necessary, but it worked for me).

```
git pull --recurse-submodules
make -C examples/... clean
make -C examples/... all
make -C examples/... erase_flash
make -C examples/... flash
```

choosegit, 2021-01-19:
I also have the this problem,In debugging，When I find something wrong, the log is：
```
>>> HomeKit: Configuring mDNS
>>> HomeKit: [Client 55] Successfully paired
>>> homekit_client_process: [Client 55] Finished processing
>>> HomeKit: [Client 55] Closing client connection
```
When there is no problem, the log is：
```
>>> HomeKit: [Client 4] Successfully paired
>>> HomeKit: [Client 4] Closing client connection
>>> HomeKit: Got new client connection: 4
>>> HomeKit: [Client 4] Pair Verify Step 1/2
>>> HomeKit: [Client 4] Pair Verify Step 2/2
>>> HomeKit: [Client 4] Found pairing with E4D1A879-C96A-4297-A752
>>> HomeKit: [Client 4] Verification successful, secure session established
>>> HomeKit: [Client 4] Get Accessories
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Update Characteristics
>>> HomeKit: [Client 4] Get Characteristics
```
It seems to have some problem with validation

maximkulkin, 2021-01-20:
@choosegit could you please recompile firmware with HomeKit debug logs enabled and reupload logs

choosegit, 2021-01-20:
`
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x40100000, len 7040, room 16 
tail 0
chksum 0xe5
load 0x3ffe8408, len 24, room 8 
tail 0
chksum 0x6c
load 0x3ffe8420, len 3332, room 8 
tail 12
chksum 0x5c
csum 0x5c
[0;32mI (41) boot: ESP-IDF v3.0.1-41-gda8a6840-dirty 2nd stage bootloader[0m
[0;32mI (41) boot: compile time 10:04:26[0m
[0;32mI (42) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (50) boot: SPI Speed      : 26.7MHz[0m
[0;32mI (57) boot: SPI Mode       : QIO[0m
[0;32mI (63) boot: SPI Flash Size : 4MB[0m
[0;32mI (69) boot: Partition Table:[0m
[0;32mI (74) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (86) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (97) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (109) boot:  2 factory          factory app      00 00 00010000 000f0000[0m
[0;32mI (120) boot: End of partition table[0m
[0;32mI (127) esp_image: segment 0: paddr=0x00010010 vaddr=0x40210010 size=0x85cec (548076) map[0m
[0;32mI (361) esp_image: segment 1: paddr=0x00095d04 vaddr=0x40295cfc size=0x1dd90 (122256) map[0m
[0;32mI (411) esp_image: segment 2: paddr=0x000b3a9c vaddr=0x3ffe8000 size=0x00bd0 (  3024) load[0m
[0;32mI (413) esp_image: segment 3: paddr=0x000b4674 vaddr=0x40100000 size=0x00080 (   128) load[0m
[0;32mI (422) esp_image: segment 4: paddr=0x000b46fc vaddr=0x40100080 size=0x057d4 ( 22484) load[0m
[0;32mI (444) boot: Loaded app from partition at offset 0x10000[0m
D (454) phy_init: loading PHY init data from application binary
D (466) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
D (481) nvs: nvs_open_from_partition backup_mac 0
D (483) system_api: load_backup_mac_data: failed to open NVS namespace (0x1102)
D (486) nvs: nvs_open_from_partition backup_mac 1
D (492) nvs: nvs_set_blob backup_mac_data 16
D (498) reset_reason: RTC reset 2 wakeup 0 store 2506093609, reason is 2
D (506) wdt: Enable task watch dog panic, panic time parameter is 11
D (515) wdt: task watch dog trigger time parameter is 15
D (520) event: running task for loop 0x40107614
D (525) event: created task for loop 0x40107614
D (531) event: created event loop 0x40107614
[0;32mI (538) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
D (551) nvs: nvs_open_from_partition backup_mac 0
D (558) nvs: nvs_get_str_or_blob backup_mac_data
D (565) nvs: nvs_close 2
[0;32mI (568) system_api: Base MAC address is not set, read default base MAC address from EFUSE[0m
D (582) nvs: nvs_open_from_partition backup_mac 0
D (589) nvs: nvs_get_str_or_blob backup_mac_data
D (596) nvs: nvs_close 3
D (599) phy_init: loading PHY init data from application binary
D (607) nvs: nvs_open_from_partition phy 0
D (613) phy_init: esp_phy_load_cal_data_from_nvs: failed to open NVS namespace (0x1102)
[0;33mW (625) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration[0m
phy_version: 1163.0, 665d56c, Jun 24 2020, 10:00:08, RTOS new
[0;32mI (814) phy_init: phy ver: 1163_0[0m
[0;33mW (815) phy_init: saving new calibration data because of checksum failure, mode(2)[0m
D (823) nvs: nvs_open_from_partition phy 1
D (830) nvs: nvs_set_blob cal_data 128
D (837) nvs: nvs_set_blob dc_table 500
D (844) nvs: nvs_close 4
D (848) nvs: nvs_open_from_partition misc 1
D (851) nvs: nvs_open_from_partition nvs.net80211 1
D (858) nvs: nvs_get opmode 1
D (861) nvs: nvs_set opmode 1 2
D (866) nvs: nvs_get_str_or_blob sta.ssid
D (871) nvs: nvs_set_blob sta.ssid 36
D (878) nvs: nvs_get_str_or_blob sta.mac
D (882) nvs: nvs_set_blob sta.mac 6
D (888) nvs: nvs_get sta.authmode 1
[... 217 lines cut ...]
D (2098) tcpip_adapter: if0 start ip lost tmr: enter
D (2105) tcpip_adapter: if0 start ip lost tmr: no need start because netif=0x401079c4 interval=120 ip=0
D (2122) tcpip_adapter: dhcp client start successfully
D (2126) event: running post SYSTEM_EVENT:4 with handler 0x40231714 on loop 0x401076f4
D (2616) tcpip_adapter: dhcpc time(ms): 0
D (3027) tcpip_adapter: if0 dhcpc cb
D (3029) tcpip_adapter: if0 ip changed=1
D (3030) system_event: SYSTEM_EVENT_STA_GOT_IP, ip:192.168.0.184, mask:255.255.254.0, gw:192.168.0.1
D (3036) event: running post IP_EVENT:0 with handler 0x4025f890 on loop 0x401076f4
[0;32mI (3047) tcpip_adapter: sta ip: 192.168.0.184, mask: 255.255.254.0, gw: 192.168.0.1[0m
D (3060) event: running post SYSTEM_EVENT:8 with handler 0x40231714 on loop 0x401076f4
WiFI ready
>>> HomeKit: Starting server
>>> HomeKit: Using existing accessory ID: 62:14:3B:36:12:8B
>>> HomeKit: Configuring mDNS
>>> homekit_run_server: Staring HTTP server
D (3116) tcpip_adapter: if0 dhcpc cb
D (3117) tcpip_adapter: if120 ip unchanged
D (3119) tcpip_adapter: ip:192.168.0.184,mask:255.255.254.0,gw:192.168.0.1

D (3186) esp_timer: start timer at callback OK
D (3286) esp_timer: start timer at callback OK
`
When SPI flash is written, the system restarts.
I think When the pairing is completed, the information is not saved correctly

choosegit, 2021-01-21:
I think the compiler or SDK version caused this problem,When I want to compile with this library(https://github.com/pfalcon/esp-open-sdk.git), some resources seem to be unavailable

choosegit, 2021-02-02:
`static inline int hwrand_generate_block(uint8_t *buf, size_t len)`
This function causes the problem

jmgutierrezc1, 2022-07-28:
I do have a similar problem, everything work very well and I can pair with Homekit, the problem come if I lose power>

0:30:05.969 -> >>> [     74] HomeKit: Using existing accessory ID: 66:41:E3:25:57:20
00:30:05.969 -> >>> [     80] HomeKit: Found admin pairing with 08F9D6E6-E39C-4A1C-A379-521136694FAA, disabling pair setup
00:30:05.969 -> >>> [     89] HomeKit: Configuring MDNS
00:30:05.969 -> >>> [     92] HomeKit: Init server over
00:30:05.969 -> Free heap: 40288, HomeKit clients: 0
00:30:10.992 -> Free heap: 40336, HomeKit clients: 0

I never get the accessory to work again. the accessories do not response again on Homekit I have to reset, connect to my wifi again delete the accessory from Homekit and re-pair

mdemir4421, 2024-02-18:
worked for me.

**void remove_all_pairings() {
    homekit_storage_reset();
}**

void setup() {
    // Your setup code here, including WiFi connection and homekit setup
    **remove_all_pairings(); // Call this function to remove all pairings**
    // Continue with the rest of your setup...
}

void loop() {
    // Your loop code here
}