maximkulkin/esp-homekit-demo #412: Pairing multiple devices?

ZweiEuro, 2021-02-25:
After initially paring one device, the homekit device doesn't show up for other homekit clients. 
Looking at the code it's saying that it already as an admin pairing, can only one homekit device be paired?

maximkulkin, 2021-02-25:
Initial (admin) pairing can be done only once. Then admin controller invites other people into the Home and when they accept invitations it adds more pairings through a special pairing management API.

ZweiEuro, 2021-02-25:
Ah ok, so the admin has to be in the same network for the time being?

ZweiEuro, 2021-03-07:
How can such an invite be sent?

rottenhowler, 2021-03-08:
In Home app select Home tab at the bottom, then press Home icon in the top left corner, in the popup menu select "Home Settings",  then in "People" section click "Invite People..." and enter people emails (not sure if it needs to be Apple IDs or not).

ZweiEuro, 2021-03-08:
ok, thanks !

closed 2021-03-08 by ZweiEuro