maximkulkin/esp-homekit-demo #413: Pairing failing without error

ZweiEuro, 2021-02-25:
```
I (15428) HAP: >>> HomeKit: Starting server

I (15438) HAP: >>> HomeKit: Resetting HomeKit storage

I (15438) HAP: >>> HomeKit: Formatting HomeKit storage at 0x200000

I (15488) HAP: >>> HomeKit: Generated new accessory ID: 90:0A:4F:C8:C1:09

I (16138) HAP: >>> HomeKit: Generated new accessory key

I (16148) HAP: >>> HomeKit: Configuring mDNS

>>> homekit_run_server: Staring HTTP server
...
I (24668) HAP: >>> HomeKit: Got new client connection: 57 from 192.168.0.2

>>> homekit_client_process: [Client 57] Got 128 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 54700
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
I (24688) HAP: >>> HomeKit: [Client 57] Pair Setup Step 1/3

>>> homekit_server_on_pair_setup: Free heap: 54644
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 57] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 53992
>>> homekit_server_on_pair_setup: [Client 57] Using user-specified password: 111-11-111
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> send_tlv_response: [Client 57] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \\\xA6pN\xEC#\x09\xD6\x04\xAB\xDDV.\x17my\xF0\xA0h\xF8V\xD0\\<\x12Z\x18\xAA\xCB\xD0\xDCJ\x92\x0A\xBE\x9C\xCA\x8A\x96\xAE\xEE\xD1L&c[\x00\xB9z\xD4;\xE9\xBE,\xF0jbD\x99\xC9\xD9\xDBm\xC0%Q\xDD2v\x0F\xBF!\x9A\x9E\x07Yb\x95\xE74\xDC\xD0^\xD6\x95\x85\x01\xDD\xDA1\xA7\xF4\xE1\xE8\xD6\xE4\x06\xDD;\x03\x1B$\xCCaC+\x0D\x81\xD1\x03\x11\x8A\x06\xC7\xE3\x1A4+?\xE9\x98E\xA8\xDD\x13\xDC\xBB\x1Eb\xB7\xED\xE9w\xEE\xD6S!\xC4\xDCT7t\x9DLjz\x0D\xEC\x10\xF3\xBC\xE5\x95\x8D\x11\x1F\x9DC,V\xC9\x03*\xCD\x81\xB4K\x00\xC4\xC0\xDB\xEBs\xE1\xED\x98\xAB\x1B\x96\xF8\xCC\xE9\x1D.\xB5\xBEb\xAB=T\xF2\x8D\xDE\xCF8z\xC5\xD7G\xA2m\xBB\xE9\xA2\xAF0\xB7\xC1\xE7\xD7\x9C\xBA@y[\xE3\xA7&\x05\x03\xFC\xC7Q\xAC\x17n\x8Dy\x13\x9A\xDD\x97\xCFa\xF9\xABv\xC0\x06le\x80\xCCj\xB0\xE2\x96\xD7\xD2L\xAB\x99\xF2\xCCO\x93\xB0\x81gR\xFC\x94F\xECy\xADNzo\x85o\xA5\x18\xD1(+Y\xC13=\x8B\xC9\x07\x80G\xCF\x11X@\x19Uy\xDFZ\x8D\x01?\x88\x11\xADV@\xEB\xB3\xE0H\xF8\xA3>+\x0Dk\xC5g\xB7 \xCBMLA%\xD1\x80\x86\xB9{\xAF\xC6\xCE3\xCC\xC3\xFE\xEE\xF23\x84\x80\x1DL/\x9E\xCD\xBF\xDD\xE9\xF4~o\xDF\xAA\xBAu\x9929\xB8\x1A\x8A\xD9\xF4*\xDA\xA9b/DVX=l\xA5/\xDB\xBC>\xBB\x8FE \x11\x09\x8B
>>> tlv_debug: Type 2 value (16 bytes): =\x92\\h\xD6Z\xF9;4\xB7\xFF}b9%|
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 57] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\\\xA6pN\xEC#\x09\xD6\x04\xAB\xDDV.\x17my\xF0\xA0h\xF8V\xD0\\<\x12Z\x18\xAA\xCB\xD0\xDCJ\x92\x0A\xBE\x9C\xCA\x8A\x96\xAE\xEE\xD1L&c[\x00\xB9z\xD4;\xE9\xBE,\xF0jbD\x99\xC9\xD9\xDBm\xC0%Q\xDD2v\x0F\xBF!\x9A\x9E\x07Yb\x95\xE74\xDC\xD0^\xD6\x95\x85\x01\xDD\xDA1\xA7\xF4\xE1\xE8\xD6\xE4\x06\xDD;\x03\x1B$\xCCaC+\x0D\x81\xD1\x03\x11\x8A\x06\xC7\xE3\x1A4+?\xE9\x98E\xA8\xDD\x13\xDC\xBB\x1Eb\xB7\xED\xE9w\xEE\xD6S!\xC4\xDCT7t\x9DLjz\x0D\xEC\x10\xF3\xBC\xE5\x95\x8D\x11\x1F\x9DC,V\xC9\x03*\xCD\x81\xB4K\x00\xC4\xC0\xDB\xEBs\xE1\xED\x98\xAB\x1B\x96\xF8\xCC\xE9\x1D.\xB5\xBEb\xAB=T\xF2\x8D\xDE\xCF8z\xC5\xD7G\xA2m\xBB\xE9\xA2\xAF0\xB7\xC1\xE7\xD7\x9C\xBA@y[\xE3\xA7&\x05\x03\xFC\xC7Q\xAC\x17n\x8Dy\x13\x9A\xDD\x97\xCFa\xF9\xABv\xC0\x06le\x80\xCCj\xB0\xE2\x96\xD7\xD2L\xAB\x99\xF2\xCCO\x03\x81\x93\xB0\x81gR\xFC\x94F\xECy\xADNzo\x85o\xA5\x18\xD1(+Y\xC13=\x8B\xC9\x07\x80G\xCF\x11X@\x19Uy\xDFZ\x8D\x01?\x88\x11\xADV@\xEB\xB3\xE0H\xF8\xA3>+\x0Dk\xC5g\xB7 \xCBMLA%\xD1\x80\x86\xB9{\xAF\xC6\xCE3\xCC\xC3\xFE\xEE\xF23\x84\x80\x1DL/\x9E\xCD\xBF\xDD\xE9\xF4~o\xDF\xAA\xBAu\x9929\xB8\x1A\x8A\xD9\xF4*\xDA\xA9b/DVX=l\xA5/\xDB\xBC>\xBB\x8FE \x11\x09\x8B\x02\x10=\x92\\h\xD6Z\xF9;4\xB7\xFF}b9%|\x06\x01\x02
>>> homekit_client_process: [Client 57] Finished processing
>>> homekit_client_process: [Client 57] Got 581 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 51792
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> tlv_debug: Type 3 value (384 bytes): <\xBD\x82\\*nD\x17\xB6\xD8\xA3\xAB\x86\x08V\x8C\x84 \xA8\x1FO\xCCY\xFA\x0BA\x15Fm\xA3\x08\xC0i\x0AA\xED\xBC\x8C\xEE\xB4$\xD6?\xB1\x1F\xAF'\xF0\xAA\xF4\xB2\x93\xBA\xC4yP\x0C,\xF3\x08\x9D\xBC\xB29\x0D\xB4wXV\xDA\x06H\x14\xE83\x16\x03B`e\x00\xACy\xA0\xCB\x00\xB9\xAB\x0C:K\xAA\xFB\xE2\xE4\x8Cmp;\x83t\xA2\xD6\x1A[U\x01\xD4D\xE8\xD9E\xAA\xB2\xD0\x9F\xAB\xDB5\xF9\xA4i\xBE\xFB?\x0Fg\xB2\x06\x16\xC0\xCB\xA9\xB8<F\x97\xFE\x87\xF3b\x1B\xB4\xB6]\xF0b\x8DV2<h\xFA\xC7\xA8\xB9"\xBE\x9F\xF4\x0E\x94\xDC BP&e\xE8Q\xE52\xA8\xA5\x02\x88\xEA\xC0\xD4\xE7\x8D\xC6\xB4]\xF6\xADX8\xA2\xA8\xBB\xA8\xB6\xAF\x1F\x072 )\xA54{\x0A\xEC\x05\x0B\xFE_\xC3\xB8\xC9\xF1+f\x82\xC3\x9F$~\x0Db\xD3!\xDC\xF0Db\x0B\xCEk\x93\x05\x15\x08\x80s\x86y\xF2\xEE\x00/\xCC$\xB9\x10\xC1Csu^\x9F\xA0\xC9\xAD$&\xAAax9\xEA\x81-q)\xAF7\xD6\xBC~X\x8E\x0BF\x8D\xAE\xC1\xBC$\x1F\xCE:8\xD3\xBA\xED\xF1Bs\xB6\xF8\xB2O\xD3\x87\xB3\xCC\x89\xB7\x853,\x8E j\xBE\xFAA6H\x0D\x1Dk*\xA3\x1B\xB9\xBB\x1CG\x8A6'\xC6\xDD\x9A\x9C\xA3k\xDFbP"!e\x98|\xBD)eH\x1C\x87\xED\xC9%\x9A\x8E\x06\x9D\xBA\xF0\x0D\\~\xE1\xE1i\xD1\xD4|O\x06"\xE6\xCA\x18\x0D<\xB9\xC7`\xD0\xD1\x85\x95Q\xFE\xE7\xC3\xCF
>>> tlv_debug: Type 4 value (64 bytes): yG\x9A\xDE\x10\xCA\xB4+\x16\x98p\xD1\xAD\x92A~\x1F\x15l\xBD\x1E\xF0-97Og1oc$\xB9\x95\xCD\x06\xCFLE\xD4\x02\x95;\xB8\x88\xB1\xC9\xCEc\x1C\xFC\xA6\x84\x9E\xF3\xC55X8\xFF'\xB8\xC1AK
I (28618) HAP: >>> HomeKit: [Client 57] Pair Setup Step 2/3

>>> homekit_server_on_pair_setup: Free heap: 51268
>>> homekit_server_on_pair_setup: [Client 57] Computing SRP shared secret
>>> homekit_server_on_pair_setup: Free heap: 51268
>>> homekit_server_on_pair_setup: [Client 57] Verifying peer's proof
>>> homekit_server_on_pair_setup: Free heap: 51272
>>> homekit_server_on_pair_setup: [Client 57] Generating own proof
>>> send_tlv_response: [Client 57] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x04
>>> tlv_debug: Type 4 value (64 bytes): C\x0B((\xE1!p\xF4\x12\xC8=G\xB6k\x9C\xF3<V\xEB\x98u\xCB\xFB\xE3\xE1TJ\xDC\x009!a\xCA\xCC\xA7\x12\\\xEB@*X\xD8\\\xFB\xF5\x11\xAC\x95\x0E\xE7U\xC8\x8C"\xC2\xF4\xC0vW\xE6\xA7W\xE4\xA1
>>> client_send: [Client 57] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 69\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x04\x04@C\x0B((\xE1!p\xF4\x12\xC8=G\xB6k\x9C\xF3<V\xEB\x98u\xCB\xFB\xE3\xE1TJ\xDC\x009!a\xCA\xCC\xA7\x12\\\xEB@*X\xD8\\\xFB\xF5\x11\xAC\x95\x0E\xE7U\xC8\x8C"\xC2\xF4\xC0vW\xE6\xA7W\xE4\xA1
>>> homekit_client_process: [Client 57] Finished processing
>>> homekit_client_process: [Client 57] Got 283 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 52096
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 5 value (154 bytes): O@\x08\x9D\xE6\xC8\x1C!\xF4M\x1E\xB3|4\xBAi\xD2\xB6\xF1\x18W;\x91TA\x0F\x02\xAD2^\x1D\xCF\xB45\xC7\x8E\x1AI\x80\xA0\xB6\xBC$\x18\xD9\x04\x87\xC7\xF5\xCD\xE29\xB4/\x10\x0F\xF4\xE2EU\xE8/\x93\xC7\xDD\xC88\xB9\xFBF\x9E\x9AG\x8A;\xEB0\xED\x96\xD1\x9F\xB5\xA4\x08\xD9\xC5\x0Di\xE2\x8D\x8A\xEEa\xF0\x9B\xEC\xD4D\xE9w\x1C\xF4^if\x1D\xD2\xF8\xC9p\x13\xF3\xB7\xE9L\xDD<\xA0\xCB\xCAN.\xAB\x90\x9Cp\xE5b\xA3\xEAo\xB8\xA5\x94<& \xDDi\xBCL`{G\xDFj;\xE3\x18\x87JY\x0A\x15
[... 71 lines cut ...]
>>> tlv_debug: Type 6 value (1 bytes): \x04
>>> client_send: [Client 57] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 3\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x06\x01\x04
I (38758) HAP: >>> HomeKit: [Client 57] Verification successful, secure session established

>>> homekit_client_process: [Client 57] Finished processing
>>> homekit_client_process: [Client 57] Got 81 incomming data
>>> homekit_client_process: [Client 57] Decrypting data
>>> homekit_client_process: [Client 57] Decrypted 63 bytes, available 0
Decrypted data (63 bytes): "GET /accessories HTTP/1.1\x0D\x0AHost: MasterGate._hap._tcp.local\x0D\x0A\x0D\x0A"
[... cut ...] r","pw","ev"],"ev":false,"description":"On","format":"bool","value":false}]}]}]}\x0D\x0A
>>> client_send: [Client 57] Sending payload: 0\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 57] Finished processing
I (39118) HAP: >>> HomeKit: [Client 57] Closing client connection
```

But i get this on the device(german):
">name< could not be added to home"
[image]

Restarting the device it seems to think there already exists an admin pairing, so i can't even repair without fully erasing the flash

ZweiEuro, 2021-02-25:
When erasing flash and retrying i get this tough:
```
>>> homekit_client_process: [Client 57] Got 128 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 15956
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
I (148188) HAP: >>> HomeKit: [Client 57] Pair Setup Step 1/3

>>> homekit_server_on_pair_setup: Free heap: 16144
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 57] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 15284
>>> homekit_server_on_pair_setup: [Client 57] Using user-specified password: 111-11-111
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Failed to get SRP verifier (code -2)
>>> crypto_srp_get_public_key: Calculating public key
E (148248) HAP: !!! HomeKit: [Client 57] Failed to dump SPR public key (code -2)
```

ZweiEuro, 2021-02-25:
Which seems to be this ? 
```
if (*buffer_size < 384) {
        *buffer_size = 384;
        return -2;
    }
```
https://github.com/maximkulkin/esp-homekit/blob/a4b8d3e8020c7af92d49ed7f2f35c461f1917d01/src/crypto.c#L170

maximkulkin, 2021-02-25:
The problem is this:
```
I (148188) HAP: >>> HomeKit: [Client 57] Pair Setup Step 1/3

>>> homekit_server_on_pair_setup: Free heap: 16144
```
Pairing cryptography uses a lot of RAM, ~22-23K at peak. So, you need to have that much memory in order for pairing to succeed. Error code -2 means that you run out of memory.

ZweiEuro, 2021-02-25:
Yes I read this in another issue, but the problem is not solved, here is a log without the memory error:
```
I (15338) HAP: >>> HomeKit: Starting server

I (15348) HAP: >>> HomeKit: Resetting HomeKit storage

I (15358) HAP: >>> HomeKit: Formatting HomeKit storage at 0x200000

I (15398) HAP: >>> HomeKit: Generated new accessory ID: F8:32:5D:08:3F:9A

I (16048) HAP: >>> HomeKit: Generated new accessory key

I (16058) HAP: >>> HomeKit: Configuring mDNS

>>> homekit_run_server: Staring HTTP server
I (42188) HAP: >>> HomeKit: Got new client connection: 57 from 192.168.0.2

>>> homekit_client_process: [Client 57] Got 128 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 54456
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 0 value (1 bytes): \x00
>>> tlv_debug: Type 6 value (1 bytes): \x01
I (42208) HAP: >>> HomeKit: [Client 57] Pair Setup Step 1/3

>>> homekit_server_on_pair_setup: Free heap: 54396
>>> crypto_srp_new: Initializing SRP
>>> homekit_server_on_pair_setup: [Client 57] Initializing crypto
>>> homekit_server_on_pair_setup: Free heap: 53740
>>> homekit_server_on_pair_setup: [Client 57] Using user-specified password: 111-11-111
>>> crypto_srp_init: Generating salt
>>> crypto_srp_init: Setting SRP username
>>> crypto_srp_init: Setting SRP params
>>> crypto_srp_init: Setting SRP password
>>> crypto_srp_init: Getting SRP verifier
>>> crypto_srp_init: Setting SRP verifier
>>> crypto_srp_get_public_key: Calculating public key
>>> send_tlv_response: [Client 57] Sending TLV response
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 3 value (384 bytes): \xC1\x9B\x11\xA2+Y\x8F\xFD\x16~\xD7\xC7N1\xD8\xAF\x91\xA57y\x8A\xE4\xE9\x95\x15\xDD\xF4\xB5^B9\x13\xF4m\x05\x9Em\x14\x10(.\xDA\xD29\xE4\x03:n\xB0\x83\x14X\xF4\x92\x1E\xEC\xE3v\xEF\xD2\x1B\x87\x82\x99\xC8*\xF8,\x19AL\x80\xB8\xB3\x93\xA9:\xA5/e\x1A"\xD1\xADW\x96\x1E\xD6>`\xE3"\x079S\xC9L\x83\xDA\xF8\x0E\x09\xA1R\xA6\xF0"\x8A\xC5\xBEo\x83&\xAE\x9C\xDE\x0C\x98\xD7L\xFE\xFE\xDFs_h\x8Bc\xAA\xD6\xFB\xEF\xF0\x1Ar\xCB1jTB\xA6w\xBE\x88z\xA9|\xE5\xDE\x17\xC3Hh\xA5\x8F\xAF\xAE\x90\xAC\xC8wqK\xB74h.\xC27\xD2\xAD)\xCC\x12\xE0\x95\xE1\xCE\xD0\x9F\xB6\xE5S!\x10\xD9E\xCCZ\xE2\xC9\x13U\xBCI\x809\xFCBCi+\xDEmI\xF2\xE1\x08 \xA0\xD5\xB3\x06Mi[xF\x1B\xF9\x82\xD1g\xB2\x0A9n\x17\x05\xB2x\x00\xC8S\x07\xB8\x14\xCD\xBAj\xE3_M\x00{\xEC\xEFu\xCB\xE6\x18\xDE\xC2-\x96\xF3\x83\xAB\xBC\xA3\xDA\xDDIj\x84\xA9P\xE2X\xE4}\x05\x00\xE4\xD4S][\xF6\xEB\xAF\xF9bd\xF0\xA3\xB3|_\x09|\xA7\xF9VB\xE9=\xCD\x97\xF5\xE5[9\xC8n\xD8\x8F5\x1F,Q\xAF\x0Ag\xE0}R\xA8Ne\x8E\x1A\xED\xDE\xCC\x9C\xEBy\xF8WpL\xF3\x17\xE1\x12\xC2P\xFD\x81\xAF=\x9DM A\xE3\x95JZ\x10|\xF7\xF0\xE2%\x97H\xCF\xACe\xDBbO\xDB\x83h[e\xDCqh"lX\xA5"\x0F\x05\xF3\x83\xC3\x83\xCF
>>> tlv_debug: Type 2 value (16 bytes): \x0C\xBD\x89\xD4z9\xB5\xDF\x0C\x8A\x1D\x12\xC6;6\xB0
>>> tlv_debug: Type 6 value (1 bytes): \x02
>>> client_send: [Client 57] Sending payload: HTTP/1.1 200 OK\x0D\x0AContent-Type: application/pairing+tlv8\x0D\x0AContent-Length: 409\x0D\x0AConnection: keep-alive\x0D\x0A\x0D\x0A\x03\xFF\xC1\x9B\x11\xA2+Y\x8F\xFD\x16~\xD7\xC7N1\xD8\xAF\x91\xA57y\x8A\xE4\xE9\x95\x15\xDD\xF4\xB5^B9\x13\xF4m\x05\x9Em\x14\x10(.\xDA\xD29\xE4\x03:n\xB0\x83\x14X\xF4\x92\x1E\xEC\xE3v\xEF\xD2\x1B\x87\x82\x99\xC8*\xF8,\x19AL\x80\xB8\xB3\x93\xA9:\xA5/e\x1A"\xD1\xADW\x96\x1E\xD6>`\xE3"\x079S\xC9L\x83\xDA\xF8\x0E\x09\xA1R\xA6\xF0"\x8A\xC5\xBEo\x83&\xAE\x9C\xDE\x0C\x98\xD7L\xFE\xFE\xDFs_h\x8Bc\xAA\xD6\xFB\xEF\xF0\x1Ar\xCB1jTB\xA6w\xBE\x88z\xA9|\xE5\xDE\x17\xC3Hh\xA5\x8F\xAF\xAE\x90\xAC\xC8wqK\xB74h.\xC27\xD2\xAD)\xCC\x12\xE0\x95\xE1\xCE\xD0\x9F\xB6\xE5S!\x10\xD9E\xCCZ\xE2\xC9\x13U\xBCI\x809\xFCBCi+\xDEmI\xF2\xE1\x08 \xA0\xD5\xB3\x06Mi[xF\x1B\xF9\x82\xD1g\xB2\x0A9n\x17\x05\xB2x\x00\xC8S\x07\xB8\x14\xCD\xBAj\xE3_M\x00{\xEC\xEFu\xCB\xE6\x18\xDE\xC2-\x96\x03\x81\xF3\x83\xAB\xBC\xA3\xDA\xDDIj\x84\xA9P\xE2X\xE4}\x05\x00\xE4\xD4S][\xF6\xEB\xAF\xF9bd\xF0\xA3\xB3|_\x09|\xA7\xF9VB\xE9=\xCD\x97\xF5\xE5[9\xC8n\xD8\x8F5\x1F,Q\xAF\x0Ag\xE0}R\xA8Ne\x8E\x1A\xED\xDE\xCC\x9C\xEBy\xF8WpL\xF3\x17\xE1\x12\xC2P\xFD\x81\xAF=\x9DM A\xE3\x95JZ\x10|\xF7\xF0\xE2%\x97H\xCF\xACe\xDBbO\xDB\x83h[e\xDCqh"lX\xA5"\x0F\x05\xF3\x83\xC3\x83\xCF\x02\x10\x0C\xBD\x89\xD4z9\xB5\xDF\x0C\x8A\x1D\x12\xC6;6\xB0\x06\x01\x02
>>> homekit_client_process: [Client 57] Finished processing
>>> homekit_client_process: [Client 57] Got 581 incomming data
>>> homekit_server_on_pair_setup: Pair Setup
>>> homekit_server_on_pair_setup: Free heap: 51240
>>> tlv_debug: Got following TLV values:
>>> tlv_debug: Type 6 value (1 bytes): \x03
>>> tlv_debug: Type 3 value (384 bytes): \xDE\xC7\xA3\xA5\xC8A6\x95{MK<dA\xE8\x1F\x9C\x0B\xE0I\xFCE\xB1\xBB\xF2\x9F\xA9\xF8\xB2\x83\x8D\xC6I6\x0E\xA4\xBDG\x04D\xCE\x80UJ\x81\x04y=\x03}\xC3\xB3\x9A\xF4c\x8BV\xB2\x19\xA9\xD39\xC6\x1D\xC1V\xD3\xB0\xE8\xCA\xC9\xECZx\xBC\xF4%\xD2\x040\xF4\xA9f>z+M\xCE\xDAH\xD9\x8D\x06 >\xEB\x8D.ME\xBF'DUi\xFA,\xB0Po\xD3n^\xEC\x90\xC7\xE7e\xCF\x04.\xBF\x1F+Z\xE8\x93b\x17\x1D\xE7\x94C\xD3\xBA\xDC\x16\xB7\xCA\x1E\x93\x86\xFA\xA5E\xA5\xE5WfD\xD8\xF4\x09\xAEo\x8C\xA1r\xB3E\xFF\xD4\xE0\xA0\xA4g\x9Dry\xC7\xB83\xE7~\xD7 \xBFR\xB1!\xDA\xFF\x0DN\xF1q\x88\xD6\xAE\xCD\xB7\x0C\xD1G\xFF\xB9#\x8B\xE4.\x94P[\xDD\x95we\x97(\xFBi6w\xBF\x15\xA7\x8B_\x07\x07\x84\xEF\x05\xE7\xA3\x91\xAB\xD9\xAC\xC23\xEA\xF5-\xD9\x05~KZGT\x9F\x98\x88f\x0B\xDEs\xD7\xADs\xDD\xCA\xC91i\xDB\xD9\x06\xA2\xB2\xD5\x9B\xD9\xC07\x10\xFFP4\x0E8\x9B{\x04\xB2\xC2\x03}l\x8B\x9AW\x8BH\x9C\x8C)\x11\xDD,\xE2\x99\x91\x99$\xBE\xDBo`\xFC\xEA\x [... cut ...]
>>> client_send: [Client 57] Sending payload: 17a\x0D\x0A"00000043-0000-1000-8000-0026BB765291","hidden":false,"primary":false,"characteristics":[{"aid":1,"iid":9,"type":"00000023-0000-1000-8000-0026BB765291","perms":["pr"],"description":"Name","format":"string","value":"\xBC\xB5\xFB?\x01"},{"aid":1,"iid":10,"type":"00000025-0000-1000-8000-0026BB765291","perms":["pr","pw","ev"],"ev":false,"description":"On","format":"bool","value":false}]}]}]}\x0D\x0A
>>> client_send: [Client 57] Sending payload: 0\x0D\x0A\x0D\x0A
>>> homekit_client_process: [Client 57] Finished processing
I (67068) HAP: >>> HomeKit: [Client 57] Closing client connection
```

maximkulkin, 2021-02-25:
That's a different issue: you have a non-ascii name in your lightbulb service, which (non-ascii name) is not supported.

ZweiEuro, 2021-02-25:
ah ok, thanks for the information, i found the issue

closed 2021-02-25 by ZweiEuro