maximkulkin/esp-homekit-demo #418: Failed to build esp-sdk

Kunalverma1502, 2021-04-02:
Earlier it was working completely fine.

now whenever I build esp-sdk using 

'docker build . -f esp-sdk-dockerfile -t esp-sdk'

It says

[INFO ]  Retrieving needed toolchain components' tarballs
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_expat_get[scripts/build/companion_libs/210-expat.sh@12]
[ERROR]  >>        called from: do_companion_libs_get[scripts/build/companion_libs.sh@15]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@591]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 0:55.27)
[00:55] / make[2]: *** [build] Error 1
#9 75.34 ct-ng:152: recipe for target 'build' failed
#9 75.34 make[2]: Leaving directory '/build/esp-open-sdk/crosstool-NG'
#9 75.34 ../Makefile:135: recipe for target '_toolchain' failed
#9 75.34 make[1]: Leaving directory '/build/esp-open-sdk/crosstool-NG'
#9 75.34 make[1]: *** [_toolchain] Error 2
#9 75.34 Makefile:130: recipe for target 'crosstool-NG/.built' failed
#9 75.34 make: *** [crosstool-NG/.built] Error 2
------
executor failed running [/bin/sh -c su docker -c "     git clone --recursive https://github.com/pfalcon/esp-open-sdk.git /build/esp-open-sdk ;     cd /build/esp-open-sdk ;     make STANDALONE=n ; "]: exit code: 2

ds5apm, 2021-04-02:
I recently had the same problem on Ubuntu.
Problem search resulted the build stopped because expat-2.2.10.tar.xz could not be downloaded. I downloaded it manually and the build was successful, but
Docker seems to be able to borrow a container that has already been created.

  https://github.com/pfalcon/esp-open-sdk/issues/237

Kunalverma1502, 2021-04-03:
can you mention the steps to do the same !
 @ds5apm

ds5apm, 2021-04-04:
As detailed in the link above, I don't think there is a need to explain further.
As a beginner, the sequence I followed is as follows.
            (Details can be found in /esp-open-sdk/crosstool-NG/build.log on your computer.)

1) Download the file from the address below.
https://master.dl.sourceforge.net/project/expat/expat/2.2.10/expat-2.2.10.tar.xz

2) Download "expat-2.1.0.tar.gz"
            Go to ~/esp-open-sdk/crosstool-NG/.build/tarballs folder.

3) Then rebuild " ...STANDALONE=n"

     I hope my experience will help you solve the problem.

fubehere, 2021-04-12:
btw the toolchain is looking for `expat-2.1.0.tar.gz`file. The url above is targeting to version 2.2.10.

nikomaxi, 2021-05-06:
@maximkulkin 
Any idea how to solve this while using docker? 
get the same errors. but I can't manually edit or download because the image will not be created on error.

Thanks.

camielverdult, 2021-05-07:
It's missing libexpat, adding the following between `git clone ...` and `make STANDALONE=n` in esp-sdk-dockerfile fixed it for me:

```
    cd /build/esp-open-sdk/crosstool-NG ; \
    ls ; \
    mkdir .build ; \
    cd .build ; \
    mkdir tarballs ; \
    cd tarballs ; \
    wget https://github.com/libexpat/libexpat/releases/download/R_2_1_0/expat-2.1.0.tar.gz ; \
```

camielverdult, 2021-05-07:
The same steps would work if you were to do this manually on a linux VM/machine

AchimPieters, 2021-05-18:
@camielverdult Yup, that works look [here](https://www.studiopieters.nl/esp-homekit-sdk-revised-installation/) for more information.