maximkulkin/esp-homekit-demo #419: Limit to # of STATELESS_PROGRAMMABLE_SWITCH buttons on a device?

jpasqua, 2021-04-22:
I've been working on a dynamic version of the button example to have multiple buttons on a single device (Wemos D1 Mini). I was having some difficulty so I decided to remove as much complexity as possible and make it statically configured. It works fine with one button, two buttons, and three buttons, but when I add a fourth I get an error from the Home app: "Unable to Add Accessory. Accessory is out of compliance." See screenshot. I've also included the log output below.

It doesn't seem likely to me that there is a hard limit of 3, but it does seem very likely that I'm missing something obvious. 

My stripped down code is available [here](https://www.dropbox.com/s/a69rjn9pthodv69/main.c?dl=0)

Any hints are appreciated.

[image]

```
Opening /dev/tty.usbserial-14430 at 115200bps...
pp_task_hdl : 3fff0360, prio:14, stack:512
pm_task_hdl : 3ffefc50, prio:1, stack:176
frc2_timer_task_hdl:0x3fff4310, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Mar 30 2021 18:46:26
phy ver: 273, pp ver: 8.3

DeviceSetupID = KV06
DevicePassword = 291-35-233
DeviceSerial = 2178XBG
DeviceName = JP2B
>>> wifi_config: Initializing WiFi config
!!! wifi_config: No configuration found
>>> wifi_config: Starting AP mode
>>> wifi_config: Starting DHCP server
mode : sta(48:3f:da:44:52:eb) + softAP(4a:3f:da:44:52:eb)
add if0
add if1
bcn 100
>>> wifi_config: Starting WiFi scan
>>> wifi_config: Starting DNS server
>>> wifi_config: Starting HTTP server
scandone
scandone
add 1
aid 1
station: c8:e0:eb:4a:d3:93 join, AID = 1
scandone
scandone
>>> wifi_config: Connecting to NewCloud
scandone
add 0
aid 3
cnt

connected with NewCloud, channel 4
dhcp client start...
>>> wifi_config: Connecting to NewCloud
scandone
no buf for probe, ie len 0
scandone
station: c8:e0:eb:4a:d3:93 join, AID = 1
>>> wifi_config: Connecting to NewCloud
scandone
scandone
>>> wifi_config: Connecting to NewCloud
scandone
scandone
ip:192.168.1.61,mask:255.255.255.0,gw:192.168.1.1
>>> wifi_config: Connected to WiFi network
station: c8:e0:eb:4a:d3:93 leave, AID = 1
rm match
bcn 0
del if1
mode : sta(48:3f:da:44:52:eb)
Function called without core lock
>>> HomeKit: Starting server
>>> HomeKit: Formatting HomeKit storage at 0x100000
>>> HomeKit: Generated new accessory ID: AC:DE:E8:B1:8F:2B
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=JP2B md=JP2Bpv=1.0id=AC:DE:E8:B1:8F:2Bc#=1s#=1ff=0sf=1ci=15
   sh=Fr2jpQ== Port=5556 TTL=4500
HOMEKIT_EVENT_SERVER_INITIALIZED
>>> wifi_config: Stopping HTTP server
>>> wifi_config: Stopping DNS server
>>> HomeKit: Got new client connection: 3 from 192.168.1.10
HOMEKIT_EVENT_CLIENT_CONNECTED
>>> HomeKit: [Client 3] Pair Setup Step 1/3
>>> HomeKit: [Client 3] Pair Setup Step 2/3
>>> HomeKit: [Client 3] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 2E198672-45E8-4CFA-83BD-73D0E3353799
HOMEKIT_EVENT_PAIRING_ADDED
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=JP2B md=JP2Bpv=1.0id=AC:DE:E8:B1:8F:2Bc#=1s#=1ff=0sf=0ci=15
   sh=Fr2jpQ== Port=5556 TTL=4500
>>> HomeKit: [Client 3] Successfully paired
>>> HomeKit: [Client 3] Closing client connection
HOMEKIT_EVENT_CLIENT_DISCONNECTED
>>> HomeKit: Got new client connection: 3 from 192.168.1.10
HOMEKIT_EVENT_CLIENT_CONNECTED
>>> HomeKit: [Client 3] Pair Verify Step 1/2
>>> HomeKit: [Client 3] Pair Verify Step 2/2
>>> HomeKit: [Client 3] Found pairing with 2E198672-45E8-4CFA-83BD-73D0E3353799
HOMEKIT_EVENT_CLIENT_VERIFIED
>>> HomeKit: [Client 3] Verification successful, secure session established
>>> HomeKit: [Client 3] Get Accessories
>>> HomeKit: [Client 3] Closing client connection
HOMEKIT_EVENT_CLIENT_DISCONNECTED
```

maximkulkin, 2021-04-22:
The problem in your code is that you're misinterpreting the value you pass to PROGRAMMABLE SWITCH EVENT characteristic when you define it. Looks like you're just assigning them sequentially like it is a button ID, while in reality it is a event state and should have values 0, 1 or 2. That's why when you add fourth one (with value 3), iOS considers whole accessory configuration to be invalid and refuses to pair. Change all values to zeros and it should work. For details on meaning of particular state values please check HAP specification.

jpasqua, 2021-04-22:
PERFECT! Thank you for the help.

closed 2021-04-22 by jpasqua