maximkulkin/esp-homekit-demo #420: common.mk No Such File

Fabersa, 2021-05-19:
Hi all,
I had this issue:

fxxxxxxxi@macpro esp-homekit-demo % make -C examples/temperature_sensor flash monitor                                                     
Makefile:17: /common.mk: No such file or directory
make: *** No rule to make target `/common.mk'.  Stop.

Where am I wrong?

renandw, 2021-05-20:
Checkout esp-open-rtos and set SDK_PATH environment variable pointing to it.

Fabersa, 2021-05-21:
Nothing wrong here but the error still the same...

export SDK_PATH= "/Users/fabiobersani/Desktop/dockfile/esp-open-rtos"
export ESPPORT=/dev/tty.usbserial-20

AchimPieters, 2021-05-21:
I have built a docker container with toolchain and use it on my Mac. [here is the instruction](https://www.studiopieters.nl/esp-homekit-sdk-revised-installation/) hope it will help.

Fabersa, 2021-05-26:
> I have built a docker container with toolchain and use it on my Mac. [here is the instruction](https://www.studiopieters.nl/esp-homekit-sdk-revised-installation/) hope it will help.

I've followed all the steps but after pressing the JOIN button on the wifi web interface the device starts blinking forever. I've waited long time and trying to add the device to the HomeKit but the device it's unreachable. Any suggestions?

AchimPieters, 2021-05-26:
@Fabersa  You can use eg. terminal form the Arduino IDE to see whats happening on your device. What does it say?

Fabersa, 2021-05-27:
--- user_init
--- OTABOOT VERSION: 2.1.2
mode : sta(84:f3:eb:b2:f6:f6)
add if0
--- pre_wifi_config
--- ota_read_rtc
--- count_step=3
--- count=0
--- standard ota
>>> wifi_config: Initializing WiFi config
>>> wifi_config: Found configuration, connecting to SilFaNdN3t
scandone
scandone
add 0
aid 1

connected with SilFaNdN3t, channel 1
dhcp client start...
cnt 
Fatal exception (9): 
epc1=0x40246b10
epc2=0x00000000
epc3=0x40203665
excvaddr=0x0000c113
depc=0x00000000
excsave1=0x0000c0ff
Registers:
a0 0000c0ff a1 3fff03f0 a2  4024afc0 a3  3ffeaeb4
a4  00000018 a5  0000003e a6  3ffe98b4 a7  000000ff
a8  0000007b a9  0000005a a10 00000002 a11 00000000
a12 3ffeaeb4 a13 3fff6be0 SAR 00000018

Stack: SP=0x3fff03f0
0x3fff03f0: 4024afc0 4024ba75 3ffec800 3ffec80c
0x3fff0400: 00000008 3ffec80c 00000000 00000019
0x3fff0410: 00000018 4025af46 3ffeaeb4 3ffec800
0x3fff0420: 000000f5 3ffec7b4 00000000 00000002
0x3fff0430: 0000005a 3ffeaeb4 3ffeaeb4 40204bdb
0x3fff0440: 4025a79d 3feffe00 3ffebf40 ffffffe0
0x3fff0450: 3ffec228 00000000 00000000 00000000
0x3fff0460: 00000000 00000000 00000000 00000000

Free Heap: 44912
_heap_start 0x3fff2668 brk 0x3fff7110 supervisor sp 0x40000000 sp-brk 36592 bytes
arena (total_size) 19112 fordblks (free_size) 8320 uordblocks (used_size) 10792

 ets Jan  8 2013,rst cause:1, boot mode:(3,3)

load 0x40100000, len 2292, room 16 
tail 4
chksum 0x57
load 0x3ffe8000, len 772, room 4 
tail 0
chksum 0x0b
csum 0x0b

rBoot v1.4.0 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: Big flash
rBoot Option: RTC data

Booting rom 0.
WAl⸮|⸮c⸮⸮o⸮x⸮⸮gn⸮l'o⸮n⸮xn⸮;c⸮o| b'⸮lb`sdpp_task_hdl : 3ffefb28, prio:14, stack:512
pm_task_hdl : 3ffef590, prio:1, stack:176
frc2_timer_task_hdl:0x3fff47e8, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Feb  3 2021 22:37:20
phy ver: 273, pp ver: 8.3

Fabersa, 2021-05-27:
Is there any way to upload the firmware via USB using Arduino IDE maybe?

AchimPieters, 2021-05-27:
@Fabersa look here: https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP8266-(Docker)

Fabersa, 2021-05-28:
> @Fabersa look here: https://github.com/maximkulkin/esp-homekit-demo/wiki/Build-instructions-ESP8266-(Docker)

That's why I've opened the ticket .... please read my first message...

Fabersa, 2021-05-28:
And if I point common.mk to the exact path I have this:

make: *** No rule to make target `/project/examples/temperature_sensor/temperature_sensor.c', needed by `build/program//temperature_sensor.o'.  Stop.

AchimPieters, 2021-05-28:
@Fabersa Unable to compile example #123

maximkulkin, 2021-05-31:
Just rechecked all instructions and rebuilt and flashed temperature_sensor example and it all worked as it should.

closed 2022-06-21 by maximkulkin