maximkulkin/esp-homekit-demo #422: Restarting the service from the program

pr200sd, 2021-06-03:
Good day.

Is it possible to orbinize the restart of the service, without restarting esp32, for example, after erasing the data, I want to be able to pair with the phone again, now I am doing a power reset for this.

maximkulkin, 2021-06-03:
No, it is not possible at the moment. Implementation is not hard, but I couldn't find any practical use case for this.

pr200sd, 2021-06-04:
Got it, thanks for the answer. My use case is as follows: after removing the accessory on the phone and erasing the data in esp32, a new device cannot be added without restarting esp32.

rottenhowler, 2021-06-05:
So, what's the problem restart the device?

maximkulkin, 2021-06-05:
Then restart the device.

pr200sd, 2021-06-05:
There are no problems to reboot the device, but my esp is used as a gateway for a multichannel relay and controls channels via modbus, https://youtu.be/7YfrBtO3Dos?t=144
and in some cases information will be lost and the relay will be reset. In principle, I have already taken into account this point and now all states are remembered before rebooting, so there are no problems.