maximkulkin/esp-homekit-demo #423: STA start STA disconnected (/esp-homekit-demo/examples/esp32/led)

exrector, 2021-06-06:
What's the catch?

`Connecting.....
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 98:f4:ab:72:32:a8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x000d7fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 24784 bytes to 15365...
Writing at 0x00001000... (100 %)
Wrote 24784 bytes (15365 compressed) at 0x00001000 in 0.8 seconds (effective 261.2 kbit/s)...
Hash of data verified.
Compressed 816032 bytes to 520743...
Writing at 0x00010000... (3 %)
Writing at 0x0001c0da... (6 %)
Writing at 0x00027802... (9 %)
Writing at 0x00031624... (12 %)
Writing at 0x00037187... (15 %)
Writing at 0x0003c842... (18 %)
Writing at 0x00042586... (21 %)
Writing at 0x000481f1... (25 %)
Writing at 0x0004dfbc... (28 %)
Writing at 0x00053bec... (31 %)
Writing at 0x000596c7... (34 %)
Writing at 0x0005edf9... (37 %)
Writing at 0x00064575... (40 %)
Writing at 0x00069d2f... (43 %)
Writing at 0x0006f42f... (46 %)
Writing at 0x00074fa0... (50 %)
Writing at 0x0007a138... (53 %)
Writing at 0x0007f5fa... (56 %)
Writing at 0x000849a3... (59 %)
Writing at 0x0008a05a... (62 %)
Writing at 0x0008f535... (65 %)
Writing at 0x00094d88... (68 %)
Writing at 0x0009a95c... (71 %)
Writing at 0x000a018f... (75 %)
Writing at 0x000a5d71... (78 %)
Writing at 0x000ac1ad... (81 %)
Writing at 0x000b3cf2... (84 %)
Writing at 0x000bbf67... (87 %)
Writing at 0x000c2268... (90 %)
Writing at 0x000c7ce8... (93 %)
Writing at 0x000cd884... (96 %)
Writing at 0x000d2e27... (100 %)
Wrote 816032 bytes (520743 compressed) at 0x00010000 in 12.7 seconds (effective 514.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 307.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Serial port /dev/cu.usbserial-0001
Connecting........_____....._
Detecting chip type... ESP32
Running idf_monitor in directory /Users/exrector/esp-homekit-demo/examples/esp32/led
Executing "/Users/exrector/.espressif/python_env/idf4.4_py3.9_env/bin/python /Users/exrector/esp/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-0001 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /Users/exrector/esp-homekit-demo/examples/esp32/led/build/led.elf -m '/Users/exrector/.espressif/python_env/idf4.4_py3.9_env/bin/python' '/Users/exrector/esp/esp-idf/tools/idf.py'"...
--- idf_monitor on /dev/cu.usbserial-0001 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6744
load:0x40078000,len:14236
ho 0 tail 12 room 4
load:0x40080400,len:3716
0x40080400: _init at ??:?

entry 0x40080680
I (28) boot: ESP-IDF v4.4-dev-1404-gc13afea63 2nd stage bootloader
I (28) boot: compile time 00:27:10
I (28) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1b89ch (112796) map
I (153) esp_image: segment 1: paddr=0002b8c4 vaddr=3ffb0000 size=04014h ( 16404) load
I (159) esp_image: segment 2: paddr=0002f8e0 vaddr=40080000 size=00738h (  1848) load
I (161) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=931dch (602588) map
I (385) esp_image: segment 4: paddr=000c3204 vaddr=40080738 size=14160h ( 82272) load
I (419) esp_image: segment 5: paddr=000d736c vaddr=50000000 size=00010h (    16) load
I (430) boot: Loaded app from partition at offset 0x10000
I (430) boot: Disabling RNG early entropy source...
I (441) cpu_start: Pro cpu up.
I (442) cpu_start: Starting app cpu, entry point is 0x400810bc
0x400810bc: call_start_cpu1 at /Users/exrector/esp/esp-idf/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (456) cpu_start: Pro cpu start user code
I (456) cpu_start: cpu freq: 160000000
I (456) cpu_start: Application information:
I (460) cpu_start: Project name:     led
I (465) cpu_start: App version:      f8dcf29-dirty
I (470) cpu_start: Compile time:     Jun  6 2021 00:26:52
I (477) cpu_start: ELF file SHA256:  e68777cbe15557af...
I (483) cpu_start: ESP-IDF:          v4.4-dev-1404-gc13afea63
I (489) heap_init: Initializing. RAM available for dynamic allocation:
I (496) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (502) heap_init: At 3FFB82F0 len 00027D10 (159 KiB): DRAM
I (508) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (515) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (521) heap_init: At 40094898 len 0000B768 (45 KiB): IRAM
I (528) spi_flash: detected chip: generic
I (532) spi_flash: flash io: dio
W (536) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (550) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (662) wifi:wifi driver task: 3ffc0c7c, prio:23, stack:6656, core=0
I (662) system_api: Base MAC address is not set
I (662) system_api: read default base MAC address from EFUSE
I (682) wifi:wifi firmware version: 4757db5
I (682) wifi:wifi certification version: v7.0
I (682) wifi:config NVS flash: enabled
I (682) wifi:config nano formating: disabled
I (692) wifi:Init data frame dynamic rx buffer num: 32
I (692) wifi:Init management frame dynamic rx buffer num: 32
I (702) wifi:Init management short buffer num: 32
I (702) wifi:Init dynamic tx buffer num: 32
I (712) wifi:Init static rx buffer size: 1600
I (712) wifi:Init static rx buffer num: 10
I (712) wifi:Init dynamic rx buffer num: 32
I (722) wifi_init: rx ba win: 6
I (722) wifi_init: tcpip mbox: 32
I (732) wifi_init: udp mbox: 6
I (732) wifi_init: tcp mbox: 6
I (732) wifi_init: tcp tx win: 5744
I (742) wifi_init: tcp rx win: 5744
I (742) wifi_init: tcp mss: 1440
I (742) wifi_init: WiFi IRAM OP enabled
I (752) wifi_init: WiFi RX IRAM OP enabled
I (762) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (862) wifi:mode : sta (98:f4:ab:72:32:a8)
I (862) wifi:enable tsf
STA start
STA disconnected
STA disconnected
STA disconnected
STA disconnected
STA disconnected
STA disconnected
...
`

maximkulkin, 2021-06-07:
What's the code?

exrector, 2021-06-08:
esp-homekit-demo/examples/esp32/led
no modifications

maximkulkin, 2021-06-08:
Did you configure your WiFi SSID and password in menuconfig?

exrector, 2021-06-09:
I followed the instructions. except i used idf.py cmake instead of make
There are no options for changing wifi menuconfig in the instructions and I used the wifi.h file
no matter. at some point the firmware ran and a test led appeared in the homekit
but now pairing error appears: Filed with OSStatus code -6700 (kUnknownErr)😕

AchimPieters, 2022-04-10:
@exrector Copy wifi.h.sample -> wifi.h and edit it with correct WiFi SSID and password.

closed 2022-06-21 by maximkulkin