maximkulkin/esp-homekit-demo #424: Matter

Kristian8606, 2021-06-23:
Hello @maximkulkin !
Do you have plans to run esp32 with Matter?

Kristian8606, 2021-06-23:
In fact, I found instructions from developer.apple.com that explain how to activate Matter
I still don't have time to build the project. If anyone has built a Matter accessory let them share their impressions

Learn more about Matter on the website.
Setup Requirements
Matter functionality is currently a Developer Preview, current enablement requirements include:
• An iPhone or iPad running iOS/iPadOS 15.0. You will need to install a profile on it (instructions below).
• An Apple TV running tvOS 15.0. You will need to install a profile on it (instructions below). If you have multiple Apple
TVs, you will need to install the profile on all of them.
• The iPhone or iPad with the Matter enablement profile must belong to the same iCloud account as the Apple TV with
the Matter enablement profile.
• A test iCloud account is highly recommended for Matter development to avoid interference with existing homes.
Profile Installation
Install the Matter Profile on your iPhone or iPad
Note: The profile will expire automatically on January 1st, 2022.
1. Download the profile and install it on the iOS/iPadOS 15.0 device.
If necessary, email the profile or use AirDrop to transfer the profile to the iOS/iPadOS 15.0 device.
2. Restart the device.
Install the Matter Profile on your Apple TV
Note: The profile will expire automatically on January 1st, 2022.
1. Download the profile and install it on the the Apple TV running tvOS 15.0.
If necessary, use Apple Configurator 2 to transfer the profile to the Apple TV.
2. Restart the device.
Ensuring Your Matter Accessory Works with the iOS/iPadOS/tvOS Developer Preview
In order to work with the current iOS/iPadOS/tvOS 15.0 Developer Preview, your Matter accessory must have the following
settings:
• Vendor ID: 0x235A
• Product ID must be one of the following:
• Lightbulb: 0xFEFF
• Outlet: 0xFEFE
• Switch: 0xFEFD
• Door Lock: 0xFEFC
• Temperature Sensor: 0xFEFB
Configuring an M5Stack to Work with the iOS/iPadOS/tvOS Developer Preview
These instructions are specific to getting started with the (Matter-provided) "all-clusters-app" on an ESP32-based
M5Stack.
1. Clone the Connected Home over IP (CHIP) Open Source project.
2. Checkout this specific commit hash for maximum compatibility with the iOS/iPadOS/tvOS 15.0 Developer Preview:
`git checkout c5fff853***********`
3. Follow these instructions on GitHub to initialize your development environment, compile the firmware and flash your
hardware.
4. Where Open Source Matter instructions talk about accessory configuration and specifically "menuconfig" for the
ESP32-based M5Stack, please defer to the next section for iOS/iPadOS/tvOS specific configuration instructions. 

Using `idf make menuconfig` to configure the M5Stack to interact with the iOS/iPadOS/tvOS
15.0 Developer Preview
1. Navigate to "Demo"
• Ensure the "Device Type" is "M5Stack"
• Set "Rendezvous Mode" to "BLE"
2. Navigate back to the main menu (use the back arrow keys)
3. Navigate to "Component Config" → "CHIP Device Layer" → “Wi-Fi Station Options"
• Ensure you do not have any entries for SSID and password
4. Navigate to "Component Config" → "CHIP Device Layer" → "Device Identification Options"
• Ensure you are using the correct Vendor ID and a supported Product ID from the “Ensuring Your Matter Accessory
Works with the iOS/iPadOS/tvOS Developer Preview” Section.
5. Press "q" and "Y" to quit and save the settings
Configuring Multiple Hardware Accessories
1. Follow instructions 1 through 5 above for your 2nd set of accessory hardware.
2. On your hardware, Navigate to "Component Config" → Chip device layer → Testing Options.
3. Edit both the "Use Test Setup Pin Code" and "Use Test Setup discriminator" to be unique values not shared with other
Hardware. For example: change the last digit of each.
Known Issues
• Pairing
• In Beta 1, Home app supports adding a maximum of 5 Matter accessories. Adding new Matter accessories may
cause existing Matter accessories in the home to fall into No Response state.
• Pairing with an "out-of-box" Matter accessory may hang for ~60s and crash. If this happens, retry the pairing
flow.
• If the accessory was previously paired with a home and removed from the home, a factory reset is required to put
the accessory back to pairing mode.
• Accessory control
• The iOS device that initiated the pairing with the accessory may not receive accessory state updates. If this
happens, reboot the iOS device.
• Matter accessories may go to "No Response" state after paired with Home app. If this happens, try:
• Remove the accessory from home, factory reset the accessory, and add it back to home.
• If the issue persists, reboot the home hub.
• If the issue persists, remove the home hub from home and add it back.
• If the issue persists, remove the home and create a new one.
• Thread accessories in an existing home may fall back to using Bluetooth if a new home is created with the same
iCloud account when the profile is installed. If this happens, remove affected Thread accessories from the home and
re-add them.
Further issues should be reported and resolved via engagement with the Open Source Matter project.

Kristian8606, 2021-07-19:
I followed all the instructions but the home app does not detect the accessory (Matter).  Has anyone else tried?

mriksman, 2021-09-15:
So you load this onto an ESP32 module, but instead of HomeKit, it's Matter? You need to start a project/repository with your code. This is really cool to start seeing Matter programming.

Kristian8606, 2021-09-15:
Require you to create an apple developer account.  Then download the pdf file with instructions

AchimPieters, 2021-09-24:
@Kristian8606  awesome! Would she the code? I'm really curious how you did it ;)

AchimPieters, 2021-09-25:
@Kristian8606 can I send you a PM?

Kristian8606, 2021-09-25:
Yes of course

hwater, 2021-09-25:
Me too: holger@ fro.at

 - typed mobil.

https://donautics.stwst.at/stories-donautics

 - typed mobil.

https://donautics.stwst.at/stories-donautics

AchimPieters, 2021-09-26:
@Kristian8606 I checked you git, but there is no contact info.

Kristian8606, 2021-09-26:
@AchimPieters Sorry, I added information

Kristian8606, 2021-09-26:
Here are my .bin files 

[Archive.zip](https://github.com/maximkulkin/esp-homekit-demo/files/7231027/Archive.zip)

mriksman, 2022-01-19:
@Kristian8606 Have you put your source code on GitHub? I'm curious to see what the code looks like. Thanks!

Kristian8606, 2022-01-19:
@mriksman View the code in the CHIP repository.
https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32

pragadheeshrajs, 2022-02-02:
@Kristian8606 Is an Apple TV a mandatory requirement for pairing Matter accessories with an iPhone or iPad ?

vicc, 2022-03-04:
@pragadheeshrajs from the developer docs, it looks like it, since you have to install a development profile on it.

@Kristian8606 do you know if you need an M5 ESP-32 board or would something like [this](url) work?

closed 2022-06-21 by maximkulkin

Kristian8606, 2022-10-30:
https://github.com/Kristian8606/esp-matter/tree/main/examples/ws2812_led_strip

AchimPieters, 2022-10-30:
@Kristian8606 how familiar are you with matter? Or with the connecthomeip / esp-matter repro’s?

Kristian8606, 2022-10-30:
I'm not familiar enough.  I modified the main light example of the esp32c3 which has one ws2812 on board to work on an esp32 board with a ws2812 strip. 
The documentation has a good description of how to create a node with clusters and attributes.

AchimPieters, 2022-10-30:
I will have a look asap 👍

AchimPieters, 2022-11-05:
@Kristian8606 would you like to join our slack group, where @maximkulkin is also a member of?

Kristian8606, 2022-11-05:
with pleasure