maximkulkin/esp-homekit-demo #426: Makefile not found

tk97tk, 2021-10-01:
When I'm executing:

`make -C examples/esp32/led menuconfig`

form the instructions from the wiki.

I´ll get the following message:

```
Makefile:17: /common.mk: No such file or directory
make: *** No rule to make target `/common.mk'.  Stop.
```

rottenhowler, 2021-10-02:
It looks like you didn't do this step:

>> Checkout esp-open-rtos and **set SDK_PATH environment variable pointing to it**.

rottenhowler, 2021-10-02:
It looks like you didn't do this step:

> Checkout esp-open-rtos and set SDK_PATH environment variable pointing to it.

tk97tk, 2021-10-02:
I use an esp32.

rottenhowler, 2021-10-02:
Ok, so for ESP32 you should use examples in examples/esp32.. examples/temperature_sensor is an ESP8266 firmware example.