maximkulkin/esp-homekit-demo #429: Motion sensor becoming non responsive

AR-Tawil, 2021-12-25:
Hi, I'm using your motion sensor code and it works perfectly. However, I noticed that the board needs resetting every two days or it will go into a No Responsive state?!!!

AchimPieters, 2022-04-10:
@AR-Tawil without information it is difficult to say where things are going wrong?

maximkulkin, 2022-06-21:
Right, it would be nice to include some more information next time:
* which board is this
* which SDK and version did you use
* was the accessory pingable while being unresponsive
* what was the output of `dns-sd -B _hap._tcp local` (natively on Mac or through Bonjour SDK downloadable from [developer.apple.com/opensource/](https://developer.apple.com/opensource/) on Windows)
* accessory logs if available

closed 2022-06-21 by maximkulkin