maximkulkin/esp-homekit-demo #433: Cannot build example - led example

christiankoback, 2022-06-19:
In my development environment (it ia a Macbook air):

- I followed instructions here: https://docs.espressif.com/projects/esp8266-rtos-sdk/en/latest/get-started/index.html

- I have set up the various env variables:
    export PATH=$PATH:<blah>/xtensa-lx106-elf/bin
    export PATH=$PATH:<blah>/ESP8266_RTOS_SDK
    export SDK_PATH=<blah>/ESP8266_RTOS_SDK
    export IDF_PATH=<blah>/ESP8266_RTOS_SDK
    export ESPPORT=<port>

when I try to use the following command (from this library's main folder):  make -C examples/led all

I get the error:  "  make: *** No rule to make target `all'.  Stop.    "

maximkulkin, 2022-06-21:
Examples in examples/ directory are designed to be built with [esp-open-rtos](https://github.com/SuperHouse/esp-open-rtos). There is an example of how to build firmware for esp8266-rtos-sdk in [examples/esp8266-rtos/sdk/led](https://github.com/maximkulkin/esp-homekit-demo/tree/master/examples/esp8266-rtos-sdk/led) although I haven't checked it lately, it might be outdated. If it does not work, let me know.

AchimPieters, 2022-06-21:
Probably your IDF_PATH is not defined. Please see https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#setup-path-to-esp-idf.

AchimPieters, 2022-06-21:
try: "make IDF_PATH=$IDF_PATH"