maximkulkin/esp-homekit-demo #437: Not able to flash anymore

dachoeks3, 2023-01-24:
I have run this couple of years ago and everything worked fine. I have set it up now and it compiles correctly but when I try to flash I get an error about 8m argument. I have tried to change it in Makefile but then it doesn't compile. Any idea what I'm doing wrong? I have installed latest esptool.py which I suspect is the issue. 

➜  esp-homekit-demo git:(master) ✗ make -C examples/sonoff_basic flash monitor
esptool.py -p /dev/tty.usbserial-1420 --baud 115200 write_flash -fs 8m -fm dout -ff 40m \
		0x0 /Users/dacho/Documents/esp/esp-open-rtos//bootloader/firmware_prebuilt/rboot.bin 0x1000 /Users/dacho/Documents/esp/esp-open-rtos//bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/sonoff_basic.bin 
usage: esptool write_flash [-h] [--erase-all] [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress] [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]] [--ignore-flash-encryption-efuse-setting]
                           [--force] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument --flash_size/-fs: invalid choice: '8m' (choose from 'detect', 'keep', '256KB', '512KB', '1MB', '2MB', '2MB-c1', '4MB', '4MB-c1', '8MB', '16MB', '32MB', '64MB', '128MB')
make: *** [flash] Error 2
➜  esp-homekit-demo git:(master) ✗

LouisLee985, 2023-01-26:
Change 8m to 8MB?

dachoeks3, 2023-01-26:
It doesn't compile then. Gives similar error. I can still compile it with 8m then manually flash it.

dachoeks3, 2023-01-27:
Issue is old esptool version in this project.